rauc/meta-rauc #14: Missing "do_image_complete" for kernel

jovere, 2017-08-08:
This is a "repost", since I did not see a response in the mailing list:

I’m attempting to add a kernel to my rauc bundle, based off of phytec’s “linux-ti” kernel: https://git.phytec.de/meta-phytec/tree/recipes-kernel/linux/linux-ti_4.4.52-phy2.bb  However, when I add the kernel, I’m getting an error `bundle.bb depends upon non-existent task do_image_complete`.  It appears that the only class which contains `do_image_complete` is the image class, but as far as I understood, the kernel shouldn’t be a system image.  I am able to create a file with just the file system, but adding a kernel or dts file does not work.

Looking inside the bundle.bbclass file, it appears that it is using the `do_image_complete` task for fetching dependencies.  Am I supposed to create this task for my kernel somehow?  What is it looking for as a return value?

Below is my bundle.bb file:

```
inherit bundle

RAUC_KEY_FILE = "${THISDIR}/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/development-1.cert.pem"
RAUC_BUNDLE_COMPATIBLE= "277264"

RAUC_BUNDLE_SLOTS="rootfs kernel"

RAUC_SLOT_rootfs="phytec-qtdemo"
RAUC_SLOT_rootfs[fstype]="ubifs"

RAUC_SLOT_kernel="linux-ti"
RAUC_SLOT_kernel[type]="kernel"
```

ejoerns, 2017-08-11:
Thanks for your requests and sorry for replying a little late.
You can continue use the mailing list if you like, but tracing issues is a bit easier for us here ;)

Using the `do_image_complete` task for the kernel is an incorrect behavior of the bundle class. I guess we have to be a bit smarter in our dependency handling there.

I will have a look at it and check if there is a more generic solution for this.

But besides that, do you really need to place the kernel outside the rootfs? It is much easier to have a consistent setup when placing the kernel inside the rootfs.

jovere, 2017-08-11:
I was looking through the different tasks, and was going to mention do_build since they both need it ... but apparently that was already used and some kind of error was coming up (823cdd6b796ca5d1579d26c5c5d784358c983c4b).  Seems odd...

The main reason I have them separate is that's the way the build was designed from Phytec and I never thought to change it, since it was working fine.

However, I'm going to require additional recipes which do not have the do_image_complete as well.  They will be a custom firmware image for an external processor that needs to be updated when the main image is updated.

As of right now, I added the following line of code, which is obviously not ideal:
```
do_fetch[depends] = ""
```
Is there a way to check if the `do_image_complete` exists first, and if it doesn't just use the `do_build`?

jovere, 2017-08-11:
Actually, after looking at it, I replaced the `do_fetch[depends]` with the following in my `bundle` recipe after the `inherit bundle` line:
```
do_fetch[depends] = " \
        linux-ti:do_build \
        phytec-qtdemo:do_image_complete \
        "
```

It appears to do everything I require for now.

ejoerns, 2017-09-19:
I've made PR #14 which addresses this issue by generating dependency list based on the image `[type]` varflag provided.

PR #19 (2017-10-24): classes/bundle: depend on do_deploy task for kernel and bootloader
Depending on `do_image_complete` task is valid for images but
inappropriate for kernel and bootloader which do not have such a task.

This changes the dependency auto-generation in order to honor the image
`type` varflag provided and depend on the `do_deploy` task for kernel
and bootloader.

This fixes #14
closed 2017-10-24 by ejoerns