rauc/meta-rauc #17: Include arbitrary files in the bundle

yadutaf, 2017-08-23:
In a custom post-install hook embedded in a rauc bundle, I'm trying to install specific files that require specific treatment (in this specific case, the bootloader itself).

I initially tried to declare it as a "virtual" slot in ``RAUC_BUNDLE_SLOTS`` of type ``file``. I was successfully able to package an arbitrary file from the deploy dir as "somefile.img" at the root of the bundle. This also create the corresponding declaration in the manifest.

Unfortunately, the current live system does not have the corresponding virtual slot declaration in its system.conf file and fails to install the bundle, which is expected.

I solved this with a do_fetch_append (see below) In the bundle recipe, with the related code in the main slot post-install hook:

```
python do_fetch_append() {
    bootloader_name = 'bootloader'
    deploy_dir = d.expand("${DEPLOY_DIR_IMAGE}")
    bundle_dir = d.expand("${S}/bundle")

    shutil.copytree(
        os.path.join(deploy_dir, bootloader_name),
        os.path.join(bundle_dir, bootloader_name),
    )
}
```

Maybe we could design a better way to embed any arbitrary files/folder inside a rauc bundle using meta-rauc. It could look like:

```
RAUC_BUNDLE_FILES = "file1 file2 ... fileN"
```

Which would copy files from the deploy directory to the bundle.

If this looks good to you, I can work on this.

ejoerns, 2017-11-13:
Well, we definitely require such a feature somehow, especially as I also intend to support providing the manifest as a file via `SRC_URI`. This would require specifying all required deploy artifacts manually.

But in addition to files form the bundle dir we will also have to include files provided by the recipe source folder, e.g. if you add a RAUC hook script or something similar.

I am not sure yet how to do this and how to combine this. At least with the latest patches applied, I've re-enabled the possibility to user the normal bitbake fetching logic. I quick guess was automatically adding an `;unpack=bundle/` to the `SRC_URI` entries to automate adding ressources via `SRC_URI`. Could be combined with an approach like the one you proposed for the artifacts from deploy folder.

If you got a useful approach, feel free to post it and we can discuss it.

ejoerns, 2019-11-21:
I came around this issue again as I wanted to remove a project-specific hack for this exact issue.

I've had again a look into #53. I basically understand the idea of using a similar description and automation as we already have for the slot handling, but I find the syntax a little to heavy (as I find the slot syntax a little heavy, too, when moving away from the straight and simple cases).

Thus I've now prepared #102 that proposes a much simpler solution. The only drawback is that one needs to take care of adding both dependencies and filenames for the artifacts, but with this it is only one- or two-liner.

PR #102 (2019-12-03): classes: bundle: add simple mechanism for adding extra artifacts to bundle
This introduces variables `RAUC_BUNDLE_EXTRA_DEPENDS` and
`RAUC_BUNDLE_EXTRA_FILES`.

With RAUC_BUNDLE_EXTRA_FILES one specifies a space-separated list of
artifacts that are first search in DEPLOY_DIR_IMAGE and then in WORKDIR.

This allows both to add artifact via SRC_URI as well as by grabbing the
generated files from other deploy recipes.

To ensure the artifacts are placed in the deploy dir before fetching
them bundle generation one must add their recipe's name to the
RAUC_BUNDLE_EXTRA_DEPENDS variable:

    RAUC_BU [... cut]
closed 2019-12-03 by ejoerns