rauc/meta-rauc #48: sysvinit support

kgolding, 2018-06-19:
Am I right in thinking this layer only supports systemd?

ejoerns, 2018-06-21:
Well, there is no real systemd configuration for the service. What is true is that we install only the systemd service files and so on. So we skip the D-Bus service wrapper for example. But this is something one can add. And we don't have any sysv init scripts.
In 99% of the systems we make systemd is used (and I don't know any good excuse not to use it except having to maintain an older system where introducing it will reslut in major structural changes)

Potentially you can simply start `rauc service` on the target. You can append the recipe and add startup scripts. I am also open to suggestions on easing the non-systemd case from the meta-rauc layer when knowing what is actually missing.

hpatriarche, 2018-10-04:
I did some basic implementation.

do_install_append() {
    # Install startup script
    install -d ${D}${sysconfdir}/init.d/
    install -m 0755 ${WORKDIR}/rauc-mark-good.sh ${D}${sysconfdir}/init.d/
    sed -i -e 's!@BINDIR@!${bindir}!g' ${D}${sysconfdir}/init.d/*.sh
}

INITSCRIPT_PACKAGES = "${PN}"
INITSCRIPT_NAME_${PN} = "rauc-mark-good.sh"
INITSCRIPT_PARAMS_${PN}  = "defaults 80"

with rauc-mark-good.sh:
#!/bin/sh

### BEGIN INIT INFO
# Provides:             rauc mark good
# Required-Start:                                                     
# Required-Stop:                                                      
# Default-Start:        2 3 4 5                                          
# Default-Stop:         0 1 6                                            
# Short-Description:    Mark as Good
### END INIT INFO                                                        

set -e

case "$1" in
  start)
	@BINDIR@/rauc status mark-good
	;;
  stop)
	;;
  restart)
	$0 start
	;;
  *)
	echo "Usage: rauc-mark-good.sh { start | | restart }" >&2
	exit 1
	;;
esac

exit 0

ejoerns, 2018-10-04:
Isn't this what we have in https://github.com/rauc/meta-rauc/pull/51 already?

ejoerns, 2018-12-19:
I've merge #51 now and thus considere this closed as I did not get any feedback.

If #51 does not cover your needs, feel free to reopen.

closed 2018-12-19 by ejoerns