rauc/meta-rauc #57: rauc used on x86 (intel nuc)

nandra, 2018-10-29:
I have successfully added rauc necessary bits and pieces to my cusotm layer and also build image which contains 2 rootfs. I've used https://github.com/rauc/rauc/blob/master/contrib/grub.conf for experiments. When board is booted it always want to boot from SlotB (default set to 2) and not 1. Also I find out that grub.cfg is taken from /dev/sda3 (rootfs B) and not from /dev/sda2 (rootfs A). Any ideas how this could happen? Thanks.

jluebbe, 2018-11-01:
You need to install grub (and grub.conf) to a separate partition, as the grub first state embeds the location of the config. Otherwise you will always start grub from one of the rootfs partition, which means that the update is not atomic (grub will fail to load if an update to that rootfs is interrupted).

nandra, 2018-11-01:
@jluebbe actually grub is installed in fat partition but yocto somehow put grubenv to rootfs partitions (not sure if I can override that somehow). Thanks.

ejoerns, 2018-11-06:
@nandra did you succeed in installing grub.conf properly?

I have no experience with this yet, but a first look at oe-core shows that by default a grub.cfg is installed into root by the grub-bootconf recipe. As it mainly aims to create a single rootfs image this is a good default but will be useless for your case.

How do you put your grub into the fat partition? Maybe you should go the same way with the grub.cfg.

ejoerns, 2018-11-06:
@nandra Also note that (depending on your requirements) you might also have the option to use UEFI bootloader capabilities directly https://github.com/rauc/rauc/blob/master/docs/integration.rst#efi (or barebox bootloader with 'bootchooser' interface).

Installing to separate fat partitions is usally a task that one needs to handle with tools like 'wic' or 'genimage' that also allow you to create an entire disk image. The other option would be to initially boot a 'setup' system that does the necessary formatting and copies the relevant files..

nandra, 2018-11-06:
@ejoerns ah ok thanks. My requirements are only to have rauc up & running :). I have setup where wic file is created with 2 rootfs + data + small fatpartition where grub in installed. But using mechanism like you provide in link would make my life easier then. But I'm not sure in which state is yocto in that regard (could be maybe bit complicated like approach with grub + grubenv).

What you latter wrote is case for hddimg (live image which can boot or install to hdd). In this case my issue was that grub bootentries was overwritten by rauc grub.cfg (also no idea why that happened)

nandra, 2018-11-12:
@ejoerns ok I was able to put grubenv + grub.cfg to /EFI/BOOT Now everythign looks fine except one strange issue. I'be used example grub.cfg from rauc and after boot always Slot B is selected. I've tried to debug it but I really don't understand how default=2 (looking at script it's impossible) :). 

Script:
```
default=0
timeout=3

set ORDER="A B"
set A_OK=0
set B_OK=0
set A_TRY=0
set B_TRY=0
load_env

# select bootable slot
for SLOT in $ORDER; do
    if [ "$SLOT" == "A" ]; then
        INDEX=1
        OK=$A_OK
        TRY=$A_TRY
        A_TRY=1
    fi
    if [ "$SLOT" == "B" ]; then
        INDEX=2
        OK=$B_OK
        TRY=$B_TRY
        B_TRY=1
    fi
    if [ "$OK" -eq 1 -a "$TRY" -eq 0 ]; then
        default=$INDEX
        break
    fi
done

# reset booted flags
if [ "$default" -eq 0 ]; then
    if [ "$A_OK" -eq 1 -a "$A_TRY" -eq 1 ]; then
        A_TRY=0
    fi
    if [ "$B_OK" -eq 1 -a "$B_TRY" -eq 1 ]; then
        B_TRY=0
    fi
fi

save_env A_TRY B_TRY

CMDLINE="quiet loglevel=0 splash udev.log-priority=3 vt.global_cursor_default=0 systemd.log_target=null plymouth.ignore-serial-consoles console=ttyxx,115200n8 panic=60"

menuentry "Rescue" {
    linux (hd0,1)/kernel root=/dev/sda1 $CMDLINE rauc.slot=R
    initrd (hd0,1)/initramfs
}

menuentry "Slot A (OK=$A_OK TRY=$A_TRY)" {
    linux (hd0,msdos2)/boot/bzImage root=LABEL=rootA $CMDLINE rauc.slot=A
    initrd (hd0,msdos2)/boot/initramfs
}

menuentry "Slot B (OK=$B_OK TRY=$B_TRY)" {
    linux (hd0,msdos3)/boot/bzImage root=LABEL=rootB $CMDLINE rauc.slot=B
    initrd (hd0,msdos3)/boot/initramfs
}

```

Any ideas or anyone tried this script and have it working? Thanks.

ejoerns, 2018-12-19:
@nandra Seems this is still open. Did you resolve it in the meantime or is the issue still valid?

nandra, 2018-12-19:
Yes resolved by enable "test"grub module in yocto. :+1:

nandra, 2018-12-20:
@ejoerns anyway other question appear. When boot from /dev/sda2 and run rauc install <bundle> then guebenv is update but after reboot with about grub.cfg selection is till in first option (not second as it should). Any idea what can cause that?

ejoerns, 2019-02-15:
I just have a system where I set up a boot with GRUB. Thus, the first major issue I hit was also the one that you already resolved yourself here, too. However, it took me a while to realize... :)

For others: When using grub-efi recipe from yocot we need to have **two** builting modules added:

    GRUB_BUILDIN_append = " test loadenv"

maybe we should add a bit convenience for this here in form of a bbappend.

About your second issue I do not have any quick clue yet but did not even reach this point myself, yet. Thus maybe if there is an issue I will hit it too. Or did you also resolve this alreay @nandra ?

ejoerns, 2019-02-17:
Ok, too old BSP, these two became defaults in add4b2aad12c44a7701eb244ae38020fc7447b4c (OE core rev https://github.com/openembedded/openembedded-core/commit/b7c9a298801c828c06b6e7da47c35f592864ce74) but are not contained in pyro, yet.

nandra, 2019-02-17:
@ejoerns nope sorry not yet resolved. Looks like by testing that sometimes it works sometimes not (I really have no idea how to debug that why as it's stored in env and should work fine all the time ;) ). Ideas? Thanks.

romainreignier, 2019-10-24:
Grub seems like the best option. Could you share the configuration on where and how place the `grub.conf` and `grubenv` at the right place.

Thank you

ejoerns, 2019-10-24:
@romainreignier barebox should basically work on EFI / x86, too. And we currently work on closing the last gaps for making it being fully supported.

The benefit is that you get a dedicated redundancy boot framework with 'barebox [bootchooser](https://barebox.org/doc/latest/user/bootchooser.html)' there. The drawback is that it was designed for ARM and thus needs a devicetree snippet as description format. But this could be loaded from a distinct location on a disk.

However, GRUB should basically work, too, yes. I do not have a fully working example I can make public, but we can work on extending the config or providing an example bsp integration layer based on your work if you like.
Please have a lookg at https://github.com/rauc/rauc/pull/499 if that already targets some of your questions.

Regards, Enrico

romainreignier, 2019-10-24:
@ejoerns Thank you for your answer.

Indeed, after reading the RAUC documentation, I was motivated on using barebox but the integration with Yocto stopped me.

Thank you for the improvement of the documentation, it helps me to understand better where to place those files. So I guess that I have to create a new recipe which installs the `grubenv` and `grub.cfg` files and sets:
```
RPROVIDES_${PN} += "virtual/grub-bootconf"
```

So, as I understand the documentation, the `grub.conf` file should contains a script based on [this example](https://github.com/rauc/rauc/blob/98d1416519037570ffd82352999ca044fa81e68c/contrib/grub.conf).

But what about `grubenv`?
Something like this ?
```
ORDER="A B"
A_OK=0
B_OK=0
A_TRY=0
B_TRY=0
```
(Based on the example here: https://rauc.readthedocs.io/en/latest/examples.html#grub-configuration).

And about partitions. As I understand, Grub and its config files should be in a separate partition.
But when I generate images without RAUC, the kernel `bzImage` is placed on the first partition with Grub and the partition is mounted as `/boot`.

So if I understand correctly, the first partition, `sda1` should contains:
```
BOOT
 ├── bootx64.efi
 ├── grub.cfg
 └── grubenv
```

And be mounted as `/boot/EFI` in the rootfs on `sda2` which already should have a `/boot` directory containing `bzImage`.

Is that correct?
If yes, could you point me to how change the layout of the rootfs produced by Yocto (I know it is more a Yocto specific question)?

I think that it would be nice to create such BSP example. I was about to make one based on `qemux86-64` but by default with Yocto does not uses any bootloader with QEMU. So we would have to create a new script which adds some difficulties. I know that Mender has [such script](https://github.com/mendersoftware/meta-mender/blob/ebc98aba646d12900cbf363794fb107de007dc09/meta-mender-qemu/scripts/mender-qemu).

What do you think? Create a BSP using `qemu` or `genericx86-64`?

romainreignier, 2019-10-25:
I have started a project using qemu here: https://github.com/romainreignier/meta-rauc-qemux86

I have figured out that the `bzImage` should be in the `/` of the rootfs. So I install it in [the `rauc-qemu-grubconf` recipe](https://github.com/romainreignier/meta-rauc-qemux86/blob/d7133d3756f592f38a19ae0b64c1f5ef328ec809/recipes-bsp/rauc-qemu-grubconf/rauc-qemu-grubconf_0.1.bb). I find that a bit ugly but I took inspiration from [this patch](https://lists.yoctoproject.org/pipermail/yocto/2018-November/043242.html).

I do not have integrated RAUC yet but the system works in qemu using:

```
 runqemu wic nographic ovmf
```

In [the wks](https://github.com/romainreignier/meta-rauc-qemux86/blob/a7ac125cf71b6371c598b65aceec76a3875b9dfb/wic/qemux86-grub-efi.wks), the `/boot` is extracted from the rootfs to be put in a separate partition. This prevents the creation of the directory `/boot` in the rootfs partition so the boot partition is not automatically mounted. I have to find a solution because it is needed to edit the `grubenv` file.

About the [`grubenv`](https://github.com/romainreignier/meta-rauc-qemux86/blob/d7133d3756f592f38a19ae0b64c1f5ef328ec809/recipes-bsp/rauc-qemu-grubconf/files/grubenv) file, I have created it manually with:
```
grub-editenv grubenv create
```
This seems to be sufficient.

romainreignier, 2019-10-30:
Hi, I have some updates.
I have asked on #yocto IRC about how to change kernel install dir but it seems that there is an issue here so I have opened a [bug report](https://bugzilla.yoctoproject.org/show_bug.cgi?id=13613).

With the scheme showed above, it there is at least 3 partitions:
1. EFI partition with grub, grubconf and grubenv
2. rootfs0 with kernel inside
2. rootfs1 with kernel inside.

But in that case, it appears that grub cannot be updated.

Obviously, for now, I do not plan to update the bootloader in the field, but who knows...

So using the UEFI "boot chooser" seems like a better plan.
We would have at least 4 partitions:
1. boot0 with EFI enabled binary like directly the kernel or systemd-boot (or even grub-efi) and the kernel image `bzImage`
2. boot1 second version of boot0 so an update of the bootloader is possible
3. root0 just the rootfs without kernel
4. root1, second rootfs, without kernel

So, this is what I have tried to do with this [local.conf snippet](https://github.com/romainreignier/meta-rauc-qemux86/blob/07f56b046e03e610d32fafcaae922676c3ac6827/README.md#2-use-systemd-boot) and this [wks file](https://github.com/romainreignier/meta-rauc-qemux86/blob/07f56b046e03e610d32fafcaae922676c3ac6827/wic/qemux86-systemd-boot-efi.wks).

But some issues remains.

**wic** is not ready yet to generate such redundant setup.
It can populate only one partition with `bootimg-efi` image, otherwise it throws an error that the file `boot.img` already exists.
And anyway, the second boot needs a different `boot.conf` option to select the right device to mount the rootfs. And the same goes for the rootfs. It is possible to use the same rootfs as source for two partitions but the `fstab` needs to be different. Furthermore, each partition declared in the wks file is mounted by default but `rauc` does not want the second partition to be mounted while installing a new bundle.

The **efibootmgr** command shown in the docs has to be done on the real hardware as it modifies the EFI variables of the machine.

Lastly, I am still not sure how to generate a rauc bundle with the bootloader.

I know that theses issues are a bit out of scope of `meta-rauc` but are essential to deploy RAUC-ready, Yocto-based system on several targets.

How do you handle these operations on your platforms?

As a general question, how to you handle the fstab issue? The UUID of the `/boot` partition needs to be changed if installed as slot.0 or slot.1.

ejoerns, 2019-11-05:
> Indeed, after reading the RAUC documentation, I was motivated on using barebox but the integration with Yocto stopped me.

It is no in oe-core, yes. But there is meta-ptx, meta-barebox, etc. basically supporting it.

> wic is not ready yet to generate such redundant setup.

This is only required when you aim to get a full disk image. In my setups (for >2GB) I often use a steup script to populate the SD card instead of moving GBs of disk images. However if sparse support works, this would be a alternative.

I am not too familiar with wic as we often use genimage for these cases.

> Obviously, for now, I do not plan to update the bootloader in the field, but who knows...

As you might now, for some cases it is possible to update the bootloader at least atomic. And the experience showed that one should never say never.

> So using the UEFI "boot chooser" seems like a better plan.

I have exactly one system using this approach. There are severals potential drawbacks one must be aware of:
* Normal BIOS will not be able to do a fallback after the first try (where we only set BootNext)
* You normally have only one attempt per slot
* Your EFI vendor needs to assure that writing EFI variables is an atomic operation
* etc.

> Lastly, I am still not sure how to generate a rauc bundle with the bootloader.

When using meta-rauc it boils down having something like

    RAUC_BUNDLE_SLOTS += "<bootloader-slotclass>"
    RAUC_SLOT_bootloader = "<bootloader-recipename>"
    RAUC_SLOT_bootloader[type] = "boot"
    RAUC_SLOT_bootloader[file] = "<name-of-bootloader-image>"

> As a general question, how to you handle the fstab issue? The UUID of the /boot partition needs to be changed if installed as slot.0 or slot.1.

The UUID is a property of the file system and thus something one cannot rely on in an image-based update. It will be changed on each image update. Thus the only reliable and persisten 'UUID' is the partition UUID (or the label if set) that one must use instead.

romainreignier, 2019-11-06:
Hi @ejoerns 

Thank you very much for you detailed answer.

> It is no in oe-core, yes. But there is meta-ptx, meta-barebox, etc. basically supporting it.

Indeed, but as `wic` is the only tool I know to create an image, it [does not support barebox](https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/scripts/lib/wic/plugins/source/bootimg-efi.py?id=d427f58977a46f2bda4b7f30cf83793373e401d3#n28). But I might try `meta-ptx` later.

> I often use a setup script to populate the SD card instead of moving GBs of disk images.

Indeed, generating large empty partitions is not efficient.
I am interested in the script approach. On the UpBoard platform, the storage is eMMC so it is not as easy as generating the partition on the SD card from the development machine. Could you point me to how such setup script could be used to automate the process of deploying a new board with Yocto? I have no idea how to do that.

> As you might now, for some cases it is possible to update the bootloader at least atomic.

I was not aware of an atomic mean to update the bootloader. Could you point me to some documentation?

> Normal BIOS will not be able to do a fallback after the first try (where we only set BootNext)

What do you mean by *normal BIOS* ? Non-EFI BIOS?

> You normally have only one attempt per slot

As far as I understand, only Barebox provides more that one attempt (or GRUB with some more scripting that the one RAUC provides), isn't it?

> Your EFI vendor needs to assure that writing EFI variables is an atomic operation

This information may be tricky to obtain.

Sorry for all these new questions but I really appreciate your precious answers.

nicolaiunrein, 2019-11-07:
I managed to install rauc on my yocto project on x86_64. After a day of trying to integrate barebox I gave up. Using Grub now is much simpler. It is important to place the grubenv into a folder called grub on the root of the boot partition. The grub.cfg goes into /EFI/BOOT/
I also have a working wic setup aka wks in my project. Except that I manually copy my grub.cfg and grubenv to the bootpartition after deploying the image to my bootable media.

romainreignier, 2019-11-08:
Hi @nicolaiunrein 
Interesting information.
Do you have one or two partitions with Grub?
Where is placed the kernel `bzImage` ?
In the wks, do you populate the second system or let it empty for the next update?
How exactly do you copy the `grub.cfg` and `grubenv` files?
Could you share you recipe or helps me on my *demo* project here: https://github.com/romainreignier/meta-rauc-qemux86

nicolaiunrein, 2019-11-08:
Hey @romainreignier 
Interesting project of yours. I think this could help a lot of people getting started with rauc on x86 systems using yocto. Please find details to my setup below:

**My Partition layout looks as follows:**
1. /boot
2. /rootfs0
3. /rootfs1
4. /images

**My boot partition looks like this:**
```
├── bzImage
├── EFI
│   └── BOOT
│       ├── bootx64.efi
│       └── grub.cfg
├── grub
│   └── grubenv
└── microcode.cpio
```

**My grub.cfg**

```
serial --unit=0 --speed=115200 --word=8 --parity=no --stop=1
CMDLINE="panic=60 quiet"

default=0
timeout=0

set ORDER="A B"
set A_OK=0
set B_OK=0
set A_TRY=0
set B_TRY=0
load_env --file "/grub/grubenv"

# select bootable slot
for SLOT in $ORDER; do
    if [ "$SLOT" == "A" ]; then
        INDEX=0
        OK=$A_OK
        TRY=$A_TRY
        A_TRY=1
    fi
    if [ "$SLOT" == "B" ]; then
        INDEX=1
        OK=$B_OK
        TRY=$B_TRY
        B_TRY=1
    fi
    if [ "$OK" -eq 1 -a "$TRY" -eq 0 ]; then
        default=$INDEX
        break
    fi
done

# reset booted flags
if [ "$default" -eq 0 ]; then
    if [ "$A_OK" -eq 1 -a "$A_TRY" -eq 1 ]; then
        A_TRY=0
    fi
    if [ "$B_OK" -eq 1 -a "$B_TRY" -eq 1 ]; then
        B_TRY=0
    fi
fi

save_env --file "/grub/grubenv" A_TRY B_TRY

menuentry "Slot A (OK=$A_OK TRY=$A_TRY)" {
    linux /bzImage root=/dev/sda2 rootwait rootfstype=ext4 quiet splash rootwait console=ttyS0,115200 console=tty0 vt.global_cursor_default=0 vt.cur_default=0
    initrd /microcode.cpio
}

menuentry "Slot B (OK=$B_OK TRY=$B_TRY)" {
    linux /bzImage root=/dev/sda3 rootwait rootfstype=ext4 quiet splash rootwait console=ttyS0,115200 console=tty0 vt.global_cursor_default=0 vt.cur_default=0
    initrd /microcode.cpio
}
```

**
At the moment I am populating both rootfs partitions within my wks ( room for improvements). I wrote a little deploy script which deploys the generated wic (and for testing my generated bundle as well) to my bootmedia, mounts the boot partition and copies the grub.cfg and grubenv files over.

**Here is my wks file:**

```
part --source bootimg-efi --sourceparams="loader=grub-efi,initrd=microcode.cpio" --ondisk sda --label boot --active --align 1024
part --source rootfs --ondisk sda --fstype=ext4 --label A --align 1024
part --source rootfs --ondisk sda --fstype=ext4 --label B --align 1024
part --ondisk sda --fstype=ext4 --label Images --size 2G --align 1024
bootloader --timeout=15 --append="rootfstype=ext4  rootwait console=ttyS0,115200 console=tty0"
```

**And here my system.conf***

```
[system]
bootloader=grub
compatible=my-distro-x86-64

[keyring]
path=/etc/rauc/ca.cert.pem

[slot.rootfs.0]
device=/dev/sda2
type=raw
bootname=A
readonly=false

[slot.rootfs.1]
device=/dev/sda3
type=raw
bootname=B
readonly=false
```

This setup works for me and I stopped working on it after I finally managed to integrate this.
I might look into improving it next week.

Future improvements I want to look into:

1. Populate only rootfs.0 and leave rootfs.1 empty for first update
2. Integrate grub.cfg and grubenv files into the wic image somehow (maybe through yocto hooks)

romainreignier, 2019-11-08:
Thanks a lot @nicolaiunrein for sharing these. Very interesting!

If I understand correctly your setup, you cannot update the kernel. Or by adding a hook that replaces the file in `/boot/bzImage` but if it fails, your system is bricked, isn't it?

About your improvements:
> 1. Populate only rootfs.0 and leave rootfs.1 empty for first update

Should be possible with `--fixed-size` like [here](https://github.com/romainreignier/meta-rauc-qemux86/blob/07f56b046e03e610d32fafcaae922676c3ac6827/wic/qemux86-systemd-boot-efi.wks#L3).

> 2. Integrate grub.cfg and grubenv files into the wic image somehow (maybe through yocto hooks)

I had managed to copy manually in the grubconf recipe [here](https://github.com/romainreignier/meta-rauc-qemux86/blob/07f56b046e03e610d32fafcaae922676c3ac6827/recipes-bsp/rauc-qemu-grubconf/rauc-qemu-grubconf_0.1.bb#L21,L22), inspired by [this poky patch](https://lists.yoctoproject.org/pipermail/yocto/2018-November/043242.html).

 > 3. Point the root to UUIDs instead of file paths within my menu entries (IMHO this must be done after the wic image has been created though)

This is very interested for me because on my target platform (UpBoard), the BIOS has been customized so the Grub's `hd0` can either points to the internal eMMC or a USB drive if connected (see [here](https://forum.up-community.org/discussion/comment/10465/#Comment_10465)).

About your deployment script, because I am new to Yocto, I am not sure how to do that.
Is your bootmedia removable so you can access it from your development machine and you run the deply script from it?
Because I use boards with a soldered eMMC, I am not sure yet how to populate the eMMC the first time.
I think I will have to create a live USB image that `dd` the wic file, executes a script and then reboot the board.
I will have to see how to simulate this step on my QEMU demo.

ejoerns, 2021-09-24:
I will close this as it did not receive any updates for quite a long time.

We have an example implementation of meta-rauc-qemux86 covering interesting capabilities of RAUC (like atomic bootloader updates or central slot status) in https://github.com/rauc/meta-rauc-community/tree/master/meta-rauc-qemux86 that itself took some inspiration from the great work done by @romainreignier.

For the future, it would be great if we could extend this layer by the additional bootloader options that are available in https://github.com/romainreignier/meta-rauc-qemux86

closed 2021-09-24 by ejoerns

romainreignier, 2021-09-24:
That is nice that you have managed to continue the work. Well done.
I have given up on Yocto for now for my x86 target but I might try your demo on Qemu.
Thanks for your work on it.