rauc/meta-rauc #61: Missing dependency to rauc-native in do_bundle

yadutaf, 2018-12-19:
After purging the './build/tmp' directory and rebuilding from the shared state cache with Yocto 2.4, I hit a case where ``do_bundle`` is called from a sysroot where ``rauc`` is not available although ``rauc-native`` is in ``DEPENDS``

```
DEBUG: Executing shell function do_bundle
/shared/build/tmp/work/somemachine-somedistro-linux/bundle-some-image/1.0-r0/temp/run.do_bundle.31126: 113: /shared/build/tmp/work/somemachine-somedistro-linux/bundle-some-image/1.0-r0/temp/run.do_bundle.31126: /shared/build/tmp/work/somemachine-somedistro-linux/bundle-some-image/1.0-r0/recipe-sysroot-native/usr/bin/rauc: not found
WARNING: exit code 127 from a shell command.
ERROR: Function failed: do_bundle (log file is located at /shared/build/tmp/work/somemachine-somedistro-linux/bundle-some-image/1.0-r0/temp/log.do_bundle.31126)
```

After explicitly adding ``do_bundle[depends] += "rauc-native:do_populate_sysroot"`` to the recipes, the build was fine again.

This kind of explicit dependencies seems to be a usual pattern in Yocto. If this approach looks good to you, I can submit a PR.

ejoerns, 2018-12-21:
@yadutaf I gave it a try but could not reproduce it for now.

Could you run `bitbake <your-bundle-recipe> -g` and verify if the generated task-depends contains the dependencies as listed below:

```
"core-bundle-minimal.do_bundle" -> "core-bundle-minimal.do_configure"
"core-bundle-minimal.do_configure" -> "core-bundle-minimal.do_prepare_recipe_sysroot"
"core-bundle-minimal.do_prepare_recipe_sysroot" -> "rauc-native.do_populate_sysroot"
```

ejoerns, 2019-01-16:
Any news on this?

ejoerns, 2019-06-12:
No news on this and did not appear during my testing. Thus I'll close as unreproducable.

If anyone runs into it, too, feel free to reopen.

closed 2019-06-12 by ejoerns