rauc/meta-rauc #65: Unable to fetch zstd

miquelsole, 2019-01-11:
Hi, since 25 december bitbake is un unable to fetch zstd using the original recipe. 

WARNING: zstd-1.33+gitAUTOINC+f3a8bd553a-r0 do_fetch: Failed to fetch URL git://github.com/facebook/zstd.git, attempting MIRRORS if available
ERROR: zstd-1.33+gitAUTOINC+f3a8bd553a-r0 do_fetch: Fetcher failure: Unable to find revision f3a8bd553a865c59f1bd6e1f68bf182cf75a8f00 in branch master even from upstream
ERROR: zstd-1.33+gitAUTOINC+f3a8bd553a-r0 do_fetch: Fetcher failure for URL: 'git://github.com/facebook/zstd.git'. Unable to fetch URL from any source.

I checked the repository and looks like the commit is there. 

any thoughts??

Thanks, 
Miuqel

ejoerns, 2019-01-11:
Mh..  seems as if this is broken on several levels...

The sha references tag `1.3.3` not `1.33` and the tag is not part of the master branch anymore.

I did not dig that deep in their repo to know if they restructured something or what has caused this, but the apparent quick-fix is to use `nobranch=1` first of all as I now did in #66.

Another task would be to fix the versioning and maybe also bump to the latest version while at it.

ejoerns, 2019-01-16:
Fixed building on sumo, thud and master with #66 merged (master) and partly cherry-picked (sumo, thud).

Thanks for reporting!

closed 2019-01-16 by ejoerns

miquelsole, 2019-01-16:
Hi,
Thanks for your quick response and solution!