rauc/meta-rauc #68: rauc 1.0 does not build on sumo branch

danc86, 2019-01-16:
Since commit d6360eda, the rauc 1.0 recipe no longer builds on sumo:

    [...]
    Applying patch add_squashfs_to_raw_handler_v4.patch
    patching file src/update_handler.c
    Hunk #1 FAILED at 1329.
    1 out of 1 hunk FAILED -- rejects in file src/update_handler.c
    Patch add_squashfs_to_raw_handler_v4.patch does not apply (enforce with -f)
    ERROR: rauc-1.0-r0 do_patch: Function failed: patch_do_patch

danc86, 2019-01-16:
Sorry, please disregard this. I should have checked where the patch was coming from first. It is in our layer not in meta-rauc.

closed 2019-01-16 by danc86

ejoerns, 2019-01-16:
Puh, I already feared I did something wrong ;)

Hope with 1.0 you now have a patch-free RAUC again!