rauc/meta-rauc #84: Cannot add hooks file to bundle recipe

phishstang65, 2019-08-19:
Hi All - I cannot seem to get my custom hook file installed in the bundle. Below is my entire bundle recipe. If I remove my new additions "RAUC_BUNDLE_EXTRA*", then the bundle builds fine and I can manually install it on my system.  If I leave those lines there, the bundle build fails with the following:

```
File: '/usr/lib/python3.6/shutil.py', lineno: 245, function: copy
     0241:
     0242:    """
     0243:    if os.path.isdir(dst):
     0244:        dst = os.path.join(dst, os.path.basename(src))
 *** 0245:    copyfile(src, dst, follow_symlinks=follow_symlinks)
     0246:    copymode(src, dst, follow_symlinks=follow_symlinks)
     0247:    return dst
     0248:
     0249:def copy2(src, dst, *, follow_symlinks=True):
File: '/usr/lib/python3.6/shutil.py', lineno: 120, function: copyfile
     0116:
     0117:    if not follow_symlinks and os.path.islink(src):
     0118:        os.symlink(os.readlink(src), dst)
     0119:    else:
 *** 0120:        with open(src, 'rb') as fsrc:
     0121:            with open(dst, 'wb') as fdst:
     0122:                copyfileobj(fsrc, fdst)
     0123:    return dst
     0124:
Exception: FileNotFoundError: [Errno 2] No such file or directory: '/home/pclass/Desktop/abc-jetson-yocto/build/tmp/work/jetson_nano-poky-linux/jetson-nano-abc-bundle/1.0-r0/abc-uboot-fw-update-change-env.sh'

ERROR: jetson-nano-abc-bundle-1.0-r0 do_unpack: [Errno 2] No such file or directory: '/home/pclass/Desktop/abc-jetson-yocto/build/tmp/work/jetson_nano-poky-linux/jetson-nano-abc-bundle/1.0-r0/abc-uboot-fw-update-change-env.sh'
```

Here is my entire bundle script:

```
inherit bundle

IMAGE_INSTALL += "abc-u-boot-env"

RAUC_KEY_FILE = "${COREBASE}/../meta-abc/recipes-core/rauc/files/openssl-ca/dev/private/development-1.key.pem"
RAUC_CERT_FILE = "${COREBASE}/../meta-abc/recipes-core/rauc/files/openssl-ca/dev/development-1.cert.pem"

RAUC_BUNDLE_COMPATIBLE = "ABC-Jetson-Nano"

RAUC_BUNDLE_VERSION = "20190805"

RAUC_BUNDLE_SLOTS = "rootfs"

RAUC_SLOT_rootfs ?= "jetson-nano-core-image-iot"
RAUC_SLOT_rootfs[fstype] = "ext4"

RAUC_BUNDLE_HOOKS[file] ?= "abc-uboot-fw-update-change-env.sh"
RAUC_BUNDLE_HOOKS[hooks] ?= "post-install"

RAUC_BUNDLE_EXTRAS ?= "abc-u-boot-env"
RAUC_BUNDLE_EXTRA_abc-u-boot-env[file] ?= "abc-uboot-fw-update-change-env.sh"

#do_fetch_append() {
#
#    import os
#    import shutil
#
#    filename = 'abc-uboot-fw-update-change-env.sh'
#    source_dir = d.expand("${COREBASE}/../meta-abc/recipes-bsp/abc-u-boot-env/files/")
#    dest_dir = d.expand("${D}")
#
#    shutil.copy(os.path.join(source_dir,filename),os.path.join(dest_dir,filename))
#
#}
```

How do I deploy my custom hooks file to the bundle? Thank you!

ejoerns, 2019-08-20:
Hi @phishstang65, 

unfortunately I did not really get what your issue is, yet.

Where do you need the `RAUC_BUNDLE_EXTRAS` for if it works fine without?
What does `RAUC_BUNDLE_EXTRAS` do? I cannot see any reference to the code that should handle this.

Regards, Enrico

phishstang65, 2019-08-20:
Hi @ejoerns,

Thanks for the reply. I need to add to the bundle my custom hooks file ( named: abc-uboot-fw-update-change-env.sh ).

From my understanding, in order to use Yocto to get my hooks file defined in the manifest.rauc file, I need to use "RAUC_BUNDLE_HOOKS[hooks]".

When I add RAUC_BUNDLE_HOOKS[hooks] and assign my filename to it, the bundle build fails and says that it cannot find the file in the bundle tmp directory. My question is how can I add my custom hooks file to the bundle? Where should I put it ( seperate recipe )? I couldn't find a guide to this - did I miss it somewhere?

Thanks,

Phil

ejoerns, 2020-09-16:
@phishstang65 Is that question still relevant?

The solution here would have been to also add the file to `SRC_URI`, thus:

``` 
SRC_URI += "myhook.sh"
RAUC_BUNDLE_HOOKS[file] = "myhook.sh"
```

I'll extend the documentation by this and then consider this closed as the thread is quite old already and the information in it is surely obsolete. Feel free to reopen if things are still unclear.

Also note that it is `RAUC_BUNDLE_HOOKS[file]` that you have to assign your filename to and not `RAUC_BUNDLE_HOOKS[hooks]` which is only the list of per-install hooks to use.

closed 2020-09-26 by ejoerns

tr3ysmith, 2024-08-07:
FYI, with the later versions of yocto, the SRC_URI needs to be adjusted:

```
SRC_URI += "file://myhook.sh"
```

ejoerns, 2024-08-08:
@tr3ysmith Thanks for pointing this out. I guess omitting the scheme was never a valid syntax.

I have fixed the example in meta-rauc bundle.bbclass which had the same issue: https://github.com/rauc/meta-rauc/pull/338

PR #143 (2020-09-26): classes: bundle: add hook to SRC_URI in example
This file is expected to be provided by the fetcher normally, thus we
should add the required SRC_URI entry in the example.

Fixes #84