rauc/meta-rauc #86: rauc-mark-good.service: Restart loop

chrthi, 2019-09-17:
`recipes-core/rauc/files/rauc-mark-good.service` does not define the service `Type`, so it defaults to `simple`. If systemd is generally configured to restart services, this will result in a loop that continuously tries to mark the target as good. I hope I haven't yet destroyed my EEPROMs...
I suggest setting
```
[Service]
Type=oneshot
```

ejoerns, 2019-09-17:
@chrthi How do you configure systemd to default to a different Restart handling?

Anyway, I would agree that it is a candidate for `oneshot`, yes.

chrthi, 2019-10-02:
Good question. I had mistakenly assumed that my settings in `/etc/systemd/system.conf` had caused this, but there you can actually only set `DefaultRestartSec`. Each service still needs `Restart` configured individually to enable this feature, and I hadn't set that for rauc-mark-good.service, so I'm unsure what caused the restart. Maybe because my `rauc-mark-good.service` depends on other services that are configured to restart?
Anyway, `oneshot` fixed it for me; `RemainAfterExit` might also be an option.