rauc/meta-rauc #87: casync bundle support in yocto receipe

yamo-goe, 2019-09-24:
Hi,
is there support in the meta-rauc layer to create casync bundles? Unfortunately I was not able to find any documentation on how to write a proper receipe to create a casync bundle for a generated image by yocto.

Thanks in advance

ejoerns, 2019-10-07:
Hi @yamo-goe,

currently meta-rauc only provides casync as host/target tool. But it lacks some automation for creating images with casync. However, the basic Idea to create a casync bundle is to first create a 'conventional' rauc bundle and then call 'rauc convert' on it to get a 'casync bundle'.

I am not sure if this should be supported by a build system, anyway. As this involves generating the chunk store / updating an existing chunk store. But maybe one should allow this if wanted.

mschwan-phytec, 2020-07-01:
I have created a pull-request that addresses this: #129

mschwan-phytec, 2020-07-09:
For now, only some additional documentation is needed explaining how to activate the generation of casync bundles. But the functionality is there with #129 being merged.

@yamo-goe Does this solve your issue, or do you first need more documentation?

ejoerns, 2020-09-15:
PR got merged and no further feedback was given here. I consider this done.

Thx @mschwan-phytec !

closed 2020-09-15 by ejoerns