rauc/meta-rauc #98: Rauc and sysvinit

triblex, 2019-11-15:
As I understand in issues #48 and #51 there has been added support for sysvinit systems. But whenever i boot my system and run `rauc status` i get the following error:

```
rauc-Message: 08:53:08.367: rauc mark: rauc status: error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: GDBus.Error:org.freedesktop.DBus.Error.Spawn.PermissionsInvalid: The permission of the setuid helper is not correct
```

Which can be solved with running `rauc service &`. 

I've tried using `chmod` and `chgrp` to change permissions on the `de.pengutronix.rauc` file without any luck. Do i need to create a sysvinit start script that runs `rauc service` or do you how i can fix this?

ejoerns, 2019-11-15:
@triblex If I get it right, this is not an issue of the RAUC integration but of the permission setting of the `dbus-daemon-launch-helper`.

Could you check if the suid bit of this is actually set?

    ls -al /usr/libexec/dbus-daemon-launch-helper

triblex, 2019-11-15:
It was not, but after running `chmod 4754 /usr/libexec/dbus-daemon-launch-helper`

I get the new following error:
```
rauc-Message: 11:28:50.594: rauc mark: rauc status: error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: GDBus.Error:org.freedesktop.DBus.Error.Spawn.ExecFailed: Failed to execute program de.pengutronix.rauc: Permission denied
```

Running `chmod 755 /usr/share/dbus-1/system-services/de.pengutronix.rauc.service` doesn't change anything and neither does a `chgrp root:root /usr/share/dbus-1/system-services/de.pengutronix.rauc.service`

triblex, 2019-11-15:
I've created a sysvinit start script 
```
  GNU nano 4.4                                                          /etc/init.d/rauc                                                           Modified
#!/bin/bash
# chkconfig: 25 20 80
# description: Starting rauc

# Source function library.
. /etc/init.d/functions

start() {
    # code to start app comes here
    # example: daemon program_name &
    rauc service &
}

stop() {
    # code to stop app comes here
    # example: killproc program_name
    killall rauc
}

case "$1" in
    start)
       start
       ;;
    stop)
       stop
       ;;
    restart)
       stop
       start
       ;;
    status)
       rauc status
       # code to check status of app comes here
       # example: status program_name
       ;;
    *)
       echo "Usage: $0 {start|stop|status|restart}"
esac
```
This solved the issue.

closed 2019-11-27 by triblex