rauc/meta-rauc #100: Including files in slot modifies the format ext4 to ext2

stobajas, 2019-11-18:
Hi,
I'm working with Rauc 1.1 and meta-rauc (from warrior release) with a linux-intel kernel 4.19 (x86).
I'm including a file in the kernel bundle slot,
```
RAUC_SLOT_kernel = "linux-intel"
RAUC_SLOT_kernel[type] = "file"
RAUC_SLOT_kernel[file] = "bzImage"
```
The system.conf:
```
[slot.kernel.0]        
parent=rootfs.0
device=/dev/sda2       
type=ext4      
               
[slot.kernel.1]        
parent=rootfs.1
device=/dev/sda3       
type=ext4 
```
Updating the kernel slot is modifying the filesystem format from ext4 to ext2.
Is it a problem you have already faced ?

I have done many tests with other files and the same problem occurs.
If I make a tar archive of the file, the file is correctly copied into the filesystem and even Rauc tool reformats the filesystem to ensure its integrity.

Thanks for your response,
Sandra

ejoerns, 2019-11-19:
> Updating the kernel slot is modifying the filesystem format from ext4 to ext2.

I wonder how this can happen. Nothing in RAUC creates ext2.

> Is it a problem you have already faced ?

No, but your config looks actually wrong. You have an ext4 slot and try to install an image to it that is not an ext4 image but a kernel image. You either need to have a raw slot without any filesystem, or your kernel needs to be part of an ext4 file system (or a tar archive).

> I have done many tests with other files and the same problem occurs.

RAUC is not designed to install individual files to slots. Would it be a problem to install a file system image?

> If I make a tar archive of the file, the file is correctly copied into the filesystem and even Rauc tool reformats the filesystem to ensure its integrity.

Yes, this is because that is the way it is meant to be used ;)

Enrico

stobajas, 2019-11-19:
Hi,
Thanks for your response.

Imho, the documentation of RAUC_SLOT_<slotclass>[type] should be a little more documented as the term "file" can be confusing leading to a file and not image file name.

When you say "RAUC is not designed to install individual files to slots" , it is more clear :).

Sandra

closed 2019-11-19 by stobajas