rauc/meta-rauc #103: Simple reference yocto+rauc implementation

ArashJavan, 2019-11-25:
Hello dear rauc-Team,

Thanks for developing this fantastic Update-Framewrok.
I am a little bit familiar with yocto and want to try to build a mini-raspberry-pi3 image including rauc as an update-frmaework using yocto.

The only problem is that i could not find any reference implementation using RAUC not even with poky! mainly i would like to know, where an dhow to set the partition-layouts needed by RAUC and generally how to integrate rauc in a sime BSP like Poky!

It would be great if you could point me to some reference implementations! 

Thanks
Arash

ejoerns, 2019-11-25:
> It would be great if you could point me to some reference implementations!

@ArashJavan This is one of the things still on my todo list. But at the moment there is no real (publicly available) reference integration of rauc. You might find some basic approaches of this in Phytecs meta-yogurt and https://github.com/romainreignier/meta-rauc-qemux86

From the 'official' side, there is the documentation that lists some necessary steps:
https://rauc.readthedocs.io/en/latest/integration.html#yocto
Did you read that alredy?

Our planning is to provide a meta-rauc-bsp layer or whatever is will be called that provides some example configurations for common platforms, such as RPI for example.

Regards, Enrico

texierp, 2019-11-25:
@ArashJavan @ejoerns, in case you are interested in, I've made (last year), a dedicated layer for Rauc
and i.MX7 WaRP7 with a A/B scheme,  see [1].

FYI, the layer has not been tested on Zeus (planned)

Regards

Pierre-Jean

[1] - https://github.com/texierp/meta-rauc-warp7/tree/warrior

ArashJavan, 2019-11-27:
@ejoerns  @texierp 
Thanks for your advise, i tried to build rauc for rp3 with yocto, but unfortunately i get follwoing error:

`rauc-Message: 08:19:09.771: rauc mark: rauc status: error retrieving slot status via D-Bus: error creating proxy: Error calling StartServiceByName for de.pengutronix.rauc: Timeout was reached`

ejoerns, 2019-11-27:
@ArashJavan What init system do you use? systemd? SysVinit?

The integration with systemd has a far better test coverage than a 'legacy' SysVinit has.

It seems that the D-Bus activation of the rauc service failed for any reason.

To see if there is no other issue, you could strat the service manually before invoking any target command:

    rauc service

If then something like `rauc status` works, it is the activation part. If this fails, there is probably some issue with RAUC / system configuration.

ArashJavan, 2019-11-27:
@ejoerns 
Thanks for your answer. 

```
rauc service
(rauc:225): rauc-ERROR **: 08:57:31.038: failed to initialize context: Key file does not start with a group
Trace/breakpoint trap
```
Content of the rauc-folder is:

```
root@raspberrypi3:~# ls /etc/rauc/
development-1.cert.pem  system.conf
```
The certificate has been created with the script provided in rauc-repo.

Am i doung something wrong?

texierp, 2019-11-27:
@ArashJavan can you share the content of 'system.conf' ?

ArashJavan, 2019-11-27:
@texierp my system.conf :

```
# [system]
compatible=RP3
bootloader=uboot
# 
[slot.rootfs.0]
device=/dev/mmcblk0p1
type=ext4
bootname=systema
# 
[slot.rootfs.1]
device=/dev/mmcblk0p2
type=ext4
bootname=systemb
```
rauc-bundle.bb-file:

```
DESCRIPTION = "RAUC bundle generator"

inherit bundle

RAUC_BUNDLE_COMPATIBLE ?= "${MACHINE}"
RAUC_BUNDLE_DESCRIPTION = "Simple Demo Bundle for RC221"
RAUC_BUNDLE_SLOTS = "rootfs"
RAUC_SLOT_rootfs = "core-image-base"
RAUC_SLOT_rootfs[fstype] ?= "ext4"

RAUC_KEY_FILE = "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"
```
rauc_%.bbappend-file:

```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SRC_URI_append := " file://system.conf"
```

ejoerns, 2019-11-27:
@ArashJavan 

``` diff
-# [system]
+[system]
```

;)

ArashJavan, 2019-11-27:
@ejoerns @texierp Thank you both guys, rauc-servcie is now running :).
But rauc-status seems to have a problem with "fw_printenv".

```
rauc status

(rauc:213): rauc-WARNING **: 11:04:36.377: Failed getting primary slot: Failed getting primary slot: Failed to start fw_printenv: Failed to execute child process ?fw_printenv? (No such file or directory)

Compatible:  RP3
Variant:     
Booted from: rootfs.1 (/dev/mmcblk0p2)
Activated:   (null) ((null))
slot states:
  rootfs.0: class=rootfs, device=/dev/mmcblk0p1, type=ext4, bootname=systema
      state=inactive, description=, parent=(none), mountpoint=(none)
      boot status=bad
  rootfs.1: class=rootfs, device=/dev/mmcblk0p2, type=ext4, bootname=systemb
      state=booted, description=, parent=(none), mountpoint=(none)
      boot status=bad
```
As far as i know it is a part of u-boot project, but i could not see where in meta-rauc the option to build the optional applications for u-boot is enabled! Isn't is necessary?  Shoud i ignore the warning of "rauc- status"?

ejoerns, 2019-11-27:
@ArashJavan You can ignore but it will surely turn into an error if you attempt installing something ;)

Switching boot sources is a bit of... a fundamental thing in redundant boot...

We have a bit of magic here, but in the end it depends on your system.conf what the right target tool to use is.

https://github.com/rauc/meta-rauc/blob/master/recipes-core/rauc/rauc-target.inc#L49

ArashJavan, 2019-11-27:
@ejoerns oh my bad, i am using the rauc-meta in branch warrior, where the u-boot options do not exist... Thanks

ejoerns, 2019-11-28:
There it just looks a little different:

https://github.com/rauc/meta-rauc/blob/warrior/recipes-core/rauc/rauc-target.inc#L49

ejoerns, 2021-09-24:
I think we can now also finally close the initial request:

For some weeks we now have https://github.com/rauc/meta-rauc-community providing some demo layers for common hardware.

It still does not support too many platforms, but I feel confident that further will be added by the community soon :smirk:

closed 2021-09-24 by ejoerns

junaidh12, 2022-12-31:
drtg