rauc/meta-rauc #122: How to add post install hook

adimhatre, 2020-04-27:
Hi, I want to add post install hook.
I have added following lines in bundle recipe, but that did not worked....
SRC_URI += " file://demo.sh"
RAUC_BUNDLE_HOOKS[file] ?= "demo.sh"
RAUC_BUNDLE_HOOKS[hooks] ?= "post-install"

Can you please help, how to execute custom script as post-install-hook....

ejoerns, 2020-04-27:
Hi @adimhatre 

A global post-install hook does not exist (yet) (https://github.com/rauc/rauc/issues/342). You can either add a per-slot post-install hook (https://rauc.readthedocs.io/en/latest/using.html#slot-hooks) or a post-install handler (https://rauc.readthedocs.io/en/latest/using.html#system-based-customization-handlers). Or extend RAUC to support post-install (bundle) hooks.

Regards, Enrico

ejoerns, 2020-09-15:
I don't think anything remained unclear here. If I should be wrong, feel free to reopen this but I'll close it for now.

closed 2020-09-15 by ejoerns