rauc/meta-rauc #125: Warnings building rauc-native on zeus

mcampbellizo, 2020-06-08:
When building the `rauc-native` recipe on zeus I get the warnings at the end of this issue. I spent a couple hours trying to get to the bottom of it and am stumped. The best I have is that removing  `inherit native` causes this to go away. Obviously that kills the point of the recipe, but there might be a clue in there. I'm not sure what `native.bbclass` would be causing this.

I found [this bug in upstream OpenEmbedded](http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-kernel/linux/kernel-devsrc.bb?id=fd74848c3c06901a77057ca03f7c01aff08ee34a) that is similar but not exactly the same. I'm not sure how to check for sure, but I don't think any key tasks are getting set to `noexec`. Due to either `native.bbclass` or any of the rauc recipe itself.

One last note, I am running an in-house distro that is based off Poky, so it's possible this is related to our changes, but it would still be inserting that we only see this with `rauc-native` and no other native recipes. Would love to hear if anyone has seen this before or has any thoughts before I spin up a clean build against Poky.

```
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-openssl.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-glib-2.0.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-opkg-utils.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-gcc-runtime.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-glibc.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-bash-completion.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-util-linux.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-libffi.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-libpcre.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-zlib.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-dbus.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-linux-libc-headers.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-libgcc.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-libxcrypt.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-libcap-ng.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-ncurses.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-bzip2.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-base-passwd.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-shadow-sysroot.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon [... cut ...] hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
WARNING: rauc-native-1.2-r0 do_prepare_recipe_sysroot: Manifest /workspaces/izo-yocto/build-tonos-r1/tmp/sstate-control/manifest-x86_64_x86_64-nativesdk-attr.populate_sysroot not found in tonos_r1 armv7ahf-neon armv7at2hf-neon armv7ahf-neon armv7at2hf-vfp armv7ahf-vfp armv6thf-vfp armv6hf-vfp armv5tehf-vfp armv5ehf-vfp armv5thf-vfp armv5hf-vfp allarch x86_64_x86_64-nativesdk (variant '')?
```

mcampbellizo, 2020-06-08:
I believe I've got to the bottom of this. `rauc` and `rauc-native` share a base `*.inc` file. In there the dependencies are set to the following:
```
DEPENDS = "openssl glib-2.0 glib-2.0-native"
```
This would normally be okay if using `BBCLASSEXTEND = "native"` to generate the native recipe, however, the native rauc has its own recipe, `rauc-native.bb`, that includes `rauc.inc` and inherits native.bbclass directly. The behavior of native.bbclass is subtly different in an explicit native recipe and one virtually created with `BBCLASSEXTEND`. When using `BBCLASSEXTEND` [it will automagically remap all entries in `DEPENDS` to their `*-native` counterpart](https://github.com/openembedded/openembedded-core/blob/56e211d0f3c6cb84f5982e5de00faeeed69c1912/meta/classes/native.bbclass#L160). [In the explicit recipe case this step is skipped](https://github.com/openembedded/openembedded-core/blob/56e211d0f3c6cb84f5982e5de00faeeed69c1912/meta/classes/native.bbclass#L137) and the writer of the recipe is expected to explicitly depend on the native versions of the recipes where it makes sense.

An easy and targeted fix would be to use class overrides in `rauc.inc` to take care of the native case explicitly.
```
DEPENDS = "openssl glib-2.0 glib-2.0-native"
DEPENDS_class-native = "openssl-native glib-2.0-native"
```

It's worth noting that in the `nativesdk` case [dependency remapping happens uncondtionally](https://github.com/openembedded/openembedded-core/blob/56e211d0f3c6cb84f5982e5de00faeeed69c1912/meta/classes/nativesdk.bbclass#L88) regardless of whether it is an explicit nativesdk recipe or one created with `BBCLASSEXTEND`. This is why we don't need a `DEPENDS_class-nativesdk = ...`.

If this all sounds sane I can whip up a PR for that fix tomorrow.

jluebbe, 2020-06-08:
I think the target package shouldn't need a dependency on glib-2.0-native.

mcampbellizo, 2020-06-08:
Looks like it is needed, the native glib tools are used to do the D-Bus binding code generation. With out it you get this error:
```
ERROR: rauc-1.2-r0 do_compile: oe_runmake failed
ERROR: rauc-1.2-r0 do_compile: Execution of '/workspaces/izo-yocto/build-tonos-r1/tmp/work/cortexa9t2hf-neon-modus-linux-gnueabi/rauc/1.2-r0/temp/run.do_compile.1440714' failed with exit code 1:
gdbus-codegen \
	--generate-c-code rauc-installer-generated \
	--c-namespace R \
	--interface-prefix de.pengutronix.rauc. \
	../rauc-1.2/src/rauc-installer.xml
/bin/bash: line 1: gdbus-codegen: command not found
Makefile:2397: recipe for target 'rauc-installer-generated.c' failed
make: *** [rauc-installer-generated.c] Error 127
make: *** Waiting for unfinished jobs....
WARNING: exit code 1 from a shell command.
```

ejoerns, 2020-06-08:
@mcampbellizo Yes, glib-2.0-native dependency is required for the exact reason you described. I try to have an eye on your fix tomorrow, thanks!