rauc/meta-rauc #128: busybox_%.bbappend doesn't set all configs recomended from docs

mcampbellizo, 2020-06-09:
In the [rauc docs](https://rauc.readthedocs.io/en/latest/integration.html#required-target-tools) it suggests that the following config options be enabled if using Busybox Tar:
```sh
CONFIG_FEATURE_TAR_AUTODETECT=y
CONFIG_FEATURE_TAR_LONG_OPTIONS=y
```
However, currently only `CONFIG_FEATURE_TAR_LONG_OPTIONS` is enabled in the `busybox_%.bbappend` in this layer. Should this also be enabled? I'm happy to open a PR, but wanted to check before putting in the effort.

jluebbe, 2020-06-09:
Yes, CONFIG_FEATURE_TAR_AUTODETECT=y should also be set. Actually, also the selected tar compression (if any) should be enabled. But that's difficult to auto-detect.

mcampbellizo, 2020-06-09:
Yeah, the compression types seem like the end user to figure out and they could add them with their own bbappend in their layer.

I'll open on a PR to add `CONFIG_FEATURE_TAR_AUTODETECT`.

Thanks for the input!

mcampbellizo, 2020-06-10:
So it turns out [in upstream Poky](https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/busybox/busybox/defconfig#n178) this is set in the BusyBox `defconfig`. This explains why this work currently. I personally like the idea of explicitly setting the config options that rauc depends on in a fragment and not depend on upstream's behavior to make this layer work. I'm still trying to get the time to open a PR, will try to get to it tomorrow.

PR #139 (2020-09-15): busybox: Add tar autodetect to config
The RAUC docs recommend setting `CONFIG_FEATURE_TAR_AUTODETECT=y` in
the BusyBox config. This is currently set in upstream Poky in the
default BusyBox config, however it is better to enforce this in
meta-rauc in case that changes upstream.

Fixes #128
closed 2020-09-15 by ejoerns