rauc/meta-rauc #135: Depedency glib-2.0 built without libmount on Yocto zeus/dunfell

a3f, 2020-07-20:
RAUC depends on glib being compiled with libmount to be able to set `mountdir=` correctly. This is especially needed when `/proc/mounts` lists `/dev/root` as the root file system.

The Yocto default is to include libmount unless overridden, but because Yocto used to use a meson cross file with the name `linux-$libc` (linux-gnueabi in my case), we failed the host_system == 'linux' check in glib's meson and glib was compiled without libmount.

For releases after, there is already a general meson fix in master. I have posted patches to fix glib-2.0 build on [zeus](https://lists.openembedded.org/g/openembedded-core/message/140803?p=,,,20,0,0,0::Created,,glib,20,2,0,75679062) and [dunfell](https://lists.openembedded.org/g/openembedded-core/message/140804?p=,,,20,0,0,0::Created,,glib,20,2,0,75679066). Older releases may be affected as well.

Question: Do we want to include this glib-2.0 patch in meta-rauc as well? How to avoid clash when the patch lands in oe-core? Just give it the same name?

ejoerns, 2020-09-15:
@a3f Your patch seems to have been accepted in zeus ([5acf5534](https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=5acf5534513105509e1385a8bb31bdc03a08555f)), but not in dunfell, probably because of [6a994a19](https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=6a994a19303e692970629043c39e16593e65f1b9).

Thus I'd assume we can close this as resolved?

a3f, 2020-09-16:
I guess so. I wasn't sure whether we want to carry the fix in meta-rauc as well, but users who could update to a new meta-rauc can update to a new zeus or dunfell as well.

closed 2020-09-16 by a3f