rauc/meta-rauc #166: microchip sama5d27 : error during kernel image boot

Noufalp90, 2020-12-19:
Hi 
I was trying to integrate rauc on sama5d27 board. i have created integrated rauc with yocto. following are the configuration files used.
meta-rauc branch: dunfell  
rauc_%.bbappend
```  
FILESEXTRAPATHS_prepend := "${THISDIR}/files:  "    
SRC_URI_append := "file://system.conf"      
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"  
```
system.conf
```
[system]
compatible=sama5d27-som1-ek-sd
bootloader=uboot
statusfile=/home/root/central_slot.raucs

[keyring]
path=demo.cert.pem

[slot.rootfs.0]
device=/dev/mmcblk0p2
type=ext4
bootname=A

[slot.rootfs.1]
device=/dev/mmcblk0p3
type=ext4
bootname=B
```
But am seeing following error during boot up time
```
(rauc:511): rauc-WARNING **: 12:35:14.924: Failed getting primary slot: Failed getting primary slot: fw_printenv failed with exit code: 247

rauc-Message: 12:35:15.051: rauc status: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code30: Failed marking 'rootfs.0' as good: Failed to r7

Poky (Yocto Project Reference Distro) 3.1 sama5d27-som1-ek-sd ttyS0
```

can someone please help me to bringup rauc on sama5d2 board?

Emantor, 2020-12-19:
> ```
> (rauc:511): rauc-WARNING **: 12:35:14.924: Failed getting primary slot: Failed getting primary slot: fw_printenv failed with exit code: 247
> ```

Indicates that fw_printenv does not work correctly on your device and returns an error code. Start by investigating why this returns an error. Since `fw_printenv` is an external tool, meta-rauc does not handle the integration for this.

Noufalp90, 2020-12-21:
thank u. i have managed the configuration issue. now if i run 'rauc install -d microchip-bundle-image-sama5d27-som1-ek-sd-20201221062700.raucb', am getting issue related to mouning. please find the below console logs
`
rauc-Message: 12:40:50.221: Debug log domains: 'rauc'
(rauc:556): rauc-DEBUG: 12:40:50.281: install started
(rauc:556): rauc-DEBUG: 12:40:50.282: input bundle: /home/root/microchip-bundle-image-sama5d27-som1-ek-sd-20201221062700.raucb
(rauc:556): rauc-DEBUG: 12:40:50.352: Trying to contact rauc service
installing
  0% Installing
  0% Determining slot states
 20% Determining slot states done.
 20% Checking bundle
 20% Verifying signature
 40% Verifying signature done.
 40% Checking bundle done.
100% Installing failed.
LastError: Failed mounting bundle: failed to run mount: Child process exited with code 32
Installing `/home/root/microchip-bundle-image-sama5d27-som1-ek-sd-20201221062700.raucb` failed

`

Emantor, 2020-12-21:
`mount` failing indicates either a problem with your kernel configuration or missing host tools, please review [kernel configuration](https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration) and the host tools section below that.

closed 2021-01-12 by Noufalp90

ejoerns, 2021-01-12:
@Noufalp90 so you did resolve your issue?

Noufalp90, 2021-01-12:
yes, it was kernel configuration issue.

gianghm1408, 2024-10-12:
> Hi I was trying to integrate rauc on sama5d27 board. i have created integrated rauc with yocto. following are the configuration files used. meta-rauc branch: dunfell rauc_%.bbappend
>
> ``` [...]

Hi @Noufalp90  Can you tell me how to fix this issue. Now I meet same issue with you.