rauc/meta-rauc #169: 3.15 Incompability with Kernel 4.14.78

baskanahm, 2020-12-22:
* Since dunfell branch is updated to rauc version 3.15 with eaf658677b4e140d0b46d60f337937d253a96cb7, Rauc failed to compile using linux kernel 4.14.78 while looking for flag LOOP_SET_BLOCK_SIZE
```
../rauc-1.5/src/mount.c: In function ‘r_setup_loop’:
../rauc-1.5/src/mount.c:201:25: error: ‘LOOP_SET_BLOCK_SIZE’ undeclared (first use in this function)
  looprc = ioctl(loopfd, LOOP_SET_BLOCK_SIZE, 4096);
                         ^
../rauc-1.5/src/mount.c:201:25: note: each undeclared identifier is reported only once for each function it appears in
```

ejoerns, 2020-12-22:
Ok, that is interesting as `LOOP_SET_BLOCK_SIZE` exists in `linux/loop.h` since v4.14. Which kernel header version are you using?

The officially supported kernel in 'dunfel' is v5.4. Is there any specific reason for using such an old kernel (4.14)?

baskanahm, 2020-12-22:
Because of hardware dependencies set by BSP Provider, it is mandatory to use 4.14.78 at my case. It seem that in 4.14.78 `LOOP_SET_BLOCK_SIZE` is defined at the same place as v.5.4 at `include/uapi/linux/loop.h`.

ejoerns, 2020-12-22:
Yes, thus my question was if you probably use older kernel headers (by default provided by `linux-libc-headers` package).

Note that vendors often tell using old kernels is required despite the hardware is well-supported in mainline, too. HW vendors are just normally not interested in providing software support and thus do not care too much for kernel updates, etc.

baskanahm, 2020-12-22:
It seems like a problem in setting correct dependencies for linux headers. Inspecting my build environment, even if I add `linux-libc-headers` as a dependency, `loops.h` file is missing under recipe-sysroot or recipe-sysroot-native.

ejoerns, 2021-01-30:
@baskanahm This should now be fixed with RAUC 1.5.1 that includes https://github.com/rauc/rauc/pull/673