rauc/meta-rauc #173: RAUC does not use IMAGE_ROOTFS_EXTRA_SPACE or IMAGE_OVERHEAD_FACTOR

einoj, 2021-01-29:
When building my image I use `IMAGE_ROOTFS_EXTRA_SPACE=400000` and `IMAGE_OVERHEAD_FACTOR=1`, to make sure that there is at least 400MB free space on the two root filesystems on my image.  This works for the image created by yocto which is created with the `do_image_wic` command. However the .ext4 partition included in the RAUC bundle does not get the extra space and is only 1.1G

```
NOTE: Creating symlink: tmp/work/clu_mk2-poky-linux-gnueabi/clu-base-image/1.0-r0/deploy-clu-base-image-image-complete/clu-base-image-clu-mk2.wic -> clu-base-image-clu-mk2-20210129084916.rootfs.wic
```
If I run `fdisk -l` I can see that both rootfs's have a size of 1.3G
```
$ fdisk -l deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic 
Disk deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic: 3,35 GiB, 3588227072 bytes, 7008256 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xcd0fc87a

Device                                                                                 Boot   Start     End Sectors  Size Id Type
deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic1 *       8192   69335   61144 29,9M  c W95 FAT32 (LBA)
deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic2        73728 2874573 2800846  1,3G 83 Linux
deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic3      2875392 5676237 2800846  1,3G 83 Linux
deploy-clu-base-image-image-complete/clu-base-image-clu-mk2-20210129084916.rootfs.wic4      5677056 7008255 1331200  650M 83 Linux

```
```
$ ll clu-base-image-clu-mk2.ext4 
-rw-r--r-- 1 root root 1,1G jan.   1  1970 /mnt/clu/clu-base-image-clu-mk2.ext4

```

I believe this happens because the .ext4 file is exported to ${DEPLOY_DIR_IMAGE} by `do_image_ext4 ` and the rootfs in the image files are resized by `do_image_wic` which is run after.

```
$ cat temp/log.task_order 

do_cleansstate (10726): log.do_cleansstate.10726
do_cleanall (10729): log.do_cleanall.10729
do_prepare_recipe_sysroot (10941): log.do_prepare_recipe_sysroot.10941
do_deploy_source_date_epoch (10944): log.do_deploy_source_date_epoch.10944
do_rootfs (10986): log.do_rootfs.10986
do_image_qa (21840): log.do_image_qa.21840
do_image (21845): log.do_image.21845
do_rootfs_wicenv (39147): log.do_rootfs_wicenv.39147
do_image_ext4 (39150): log.do_image_ext4.39150
do_image_wic (39155): log.do_image_wic.39155
do_image_complete (39473): log.do_image_complete.39473
```

The extra space is added when do_image_wic calls set_image_size()
```
meta-poky/meta/classes/image.bbclass:569:python set_image_size () {
meta-poky/meta/classes/image_types_wic.bbclass:145:do_rootfs_wicenv[prefuncs] = 'set_image_size'
```

Is there a way to get the 400MB extra space added to the .ext4 file in the RAUC bundle? Perhaps by using the .wic file which is also in  ${DEPLOY_DIR_IMAGE}, or by running `set_image_size ()` from image.bbclass before including the .ext4 file in the RAUC bundle?

einoj, 2021-01-29:
Looks like the IMAGE_ROOTFS_SIZE variable (which depends on the _EXTRA_SPACE and _OVERHEAD_FACTOR variables) used to be ignored by wks too https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/classes?h=dunfell&id=b404c0cd3c12958e705f666c29920816912c2d56

I suggest implementing a patch similar to the linked commit.

ejoerns, 2021-01-30:
@einoj The image size of an ext4 should not be that critical. What is critical is the *partition size*, as this can prevent you from installing larger ext4 file system images in the future.

Extra size in the file system would also only be required if the filesystem is used for writing data on the target (which might be problematic when using it for image updates). However, you can always extend the size of the ext4 to fill the entire partition. RAUC also supports this out-of-the-box with the [resize](https://rauc.readthedocs.io/en/latest/reference.html#system-configuration-file) option in the slot configuration.

Would that already fit your needs? Or what is the underlying requirement for having free space in images?

einoj, 2021-02-04:
The `resize` option sounds like exactly what I need. Do you know if I can set this option using the bundle.bbclass? For example by setting  `RAUC_SLOT_rootfs[resize] ?= "true"`?

ejoerns, 2021-02-04:
@einoj No, this option is available only in the system configuration.

From the bundle, you could probably call a post-install hook doing the same. Then you have to ensure that the slot is not mounted (anymore).

jluebbe, 2021-02-04:
If you don't require bit identical slot contents, you an also use a tar file in the bundle. RAUC will format the destination slot and extract the archive.

einoj, 2021-02-11:
Thanks for the advice! I had a closer look at this problem and setting `resize=true` in the system configuration is good enough for my project.

closed 2021-02-11 by einoj