rauc/meta-rauc #174: Setting boot limit or no of tries to 1?

amod-amatya, 2021-01-30:
It is provided we have these parameters BOOT_A_LEFT and BOOT_B_LEFT which are set to 3 in the [https://github.com/rauc/rauc/blob/master/contrib/uboot.sh](url) boot script that you have provided. I want to change this boot limit to 1, so I tried to set the above parameters as 1. But what I observe is on reboot this again changes to 3

This log is before reboot i.e on a new image flash
```
Found valid slot A, 1 attempts remaining
skip remaining slots
Saving Environment to FAT... OK
```

After reboot
```
Found valid slot A, 3 attempts remaining
skip remaining slots
Saving Environment to FAT... OK
```

Is it that a patch is needed for changing these variables on the source code itself?

ejoerns, 2021-01-30:
@amod-amatya The reset to 3 attempts is very likely done by the rauc [mark-good service](https://rauc.readthedocs.io/en/latest/using.html#react-to-a-successfully-booted-system-failed-boot). This sets the number of retries to a fixed value of '3'. You are probably intereseted in https://github.com/rauc/rauc/pull/661 which makes this configurable.

amod-amatya, 2021-01-31:
@ejoerns thank you exactly what I was looking for. thank you for making this awesome update tool

ejoerns, 2021-02-01:
@amod-amatya great to hear, thanks!

closed 2021-02-01 by ejoerns