rauc/meta-rauc #185: Yocto: No instructions for creating a bundle with a Post-install Hook

Test4564, 2021-03-23:
Hello together,

at the moment I am trying to create a bundle with a post-install hook.
But there I can not find any example for this purpose in the documentation...

That is my actual try:
My update-bundle.bb:

`DESCRIPTION = "RAUC bundle generator"

inherit bundle

RAUC_BUNDLE_COMPATIBLE = "RaspberryPi4"
RAUC_BUNDLE_VERSION = "v20200703"
RAUC_BUNDLE_DESCRIPTION = "RAUC Demo Bundle"
RAUC_BUNDLE_SLOTS = "rootfs" 
RAUC_SLOT_rootfs = "core-image-minimal"
RAUC_SLOT_rootfs[fstype] = "ext4"

RAUC_KEY_FILE = "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"

RAUC_BUNDLE_HOOKS[file] = "${THISDIR}/files/hook.sh"
RAUC_BUNDLE_HOOKS[hooks] = "post-install"`

bitbake update-bundle fails now with:
FileNotFoundError: [Errno 2] No such file or directory: '/home/xxx/poky-gatesgarth/build-dir/tmp/work/raspberrypi3-poky-linux-gnueabi/update-bundle/1.0-r0//home/xxx/poky-gatesgarth/customlayers/meta-smarthome/meta-rauc-raspberrypi/recipes-core/bundles/files/hook.sh'

The update with rauc generally works for me.
But I can not get work a hook.sh file after the update.
=> I need it to copy configuration files from the currently active slot to the new active slot

Can you give me a link to an example?
Or give me some advice what I have to do here?

Thank you very much!

Alex

ejoerns, 2021-03-24:
@Test4564 Where is your hook file currently placed in the BSP?

An option is to use simple fetcher handling here and do

```
SRC_URI += "file://hooks.sh"
RAUC_BUNDLE_HOOKS[file] = "hook.sh"
```

Test4564, 2021-03-24:
Hello ejoerns,

I started with this tutorial:
https://www.konsulko.com/getting-started-with-rauc-on-raspberry-pi-2/

So I used the meta-rauc-community layer and modified the contents of the layer for my usecase.
=> I added the hook.sh file in the recipes-core/bundles/files
=> I modified the update-bundle.bb file (see my first post)

meta-rauc-smarthome directory:
`.
├── conf
│   └── layer.conf
├── COPYING.MIT
├── meta-rauc-raspberrypi
│   ├── classes
│   │   └── rauc-integration.bbclass
│   ├── conf
│   │   └── layer.conf
│   ├── README
│   ├── recipes-bsp
│   │   └── rpi-u-boot-scr
│   │       ├── files
│   │       │   └── boot.cmd.in
│   │       └── rpi-u-boot-scr.bbappend
│   ├── recipes-core
│   │   ├── base-files
│   │   │   ├── base-files_%.bbappend
│   │   │   └── files
│   │   │       └── fstab
│   │   ├── bundles
│   │   │   ├── files
│   │   │   │   ├── development-1.cert.pem
│   │   │   │   ├── development-1.key.pem
│   │   │   │   └── hook.sh
│   │   │   └── update-bundle.bb
│   │   ├── rauc
│   │   │   ├── files
│   │   │   │   ├── ca.cert.pem
│   │   │   │   ├── rauc-grow-data-partition.service
│   │   │   │   └── system.conf
│   │   │   └── rauc_%.bbappend
│   │   └── udev
│   │       ├── files
│   │       │   └── raspberrypi-rauc.rules
│   │       └── udev-extraconf_%.bbappend
│   ├── recipes-kernel
│   │   └── linux
│   │       ├── files
│   │       │   └── rauc.cfg
│   │       └── linux-raspberrypi%.bbappend
│   └── wic
│       └── sdimage-dual-raspberrypi.wks.in
└── README`

If I add your suggested line to the update-bundle.bb file:
=> SRC_URI += "file://hooks.sh"
     RAUC_BUNDLE_HOOKS[file] = "hook.sh"

`DESCRIPTION = "RAUC bundle generator"

inherit bundle

RAUC_BUNDLE_COMPATIBLE = "RaspberryPi4"
RAUC_BUNDLE_VERSION = "v20200703"
RAUC_BUNDLE_DESCRIPTION = "RAUC Demo Bundle"
RAUC_BUNDLE_SLOTS = "rootfs" 
RAUC_SLOT_rootfs = "core-image-minimal"
RAUC_SLOT_rootfs[fstype] = "ext4"

RAUC_KEY_FILE = "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"

SRC_URI += "file://hook.sh"
RAUC_BUNDLE_HOOKS[file] = "hook.sh"
RAUC_BUNDLE_HOOKS[hooks] = "post-install"`

Then I have a license error:
`ERROR: update-bundle-1.0-r0 do_populate_lic: QA Issue: update-bundle: Recipe file fetches files and does not have license file information (LIC_FILES_CHKSUM) [license-checksum]
ERROR: update-bundle-1.0-r0 do_populate_lic: Fatal QA errors found, failing task.
ERROR: Logfile of failure stored in: /home/alexander/poky-gatesgarth/greencube-build/tmp/work/raspberrypi3-poky-linux-gnueabi/update-bundle/1.0-r0/temp/log.do_populate_lic.533
ERROR: Task (/home/alexander/poky-gatesgarth/customlayers/meta-smarthome/meta-rauc-raspberrypi/recipes-core/bundles/update-bundle.bb:do_populate_lic) failed with exit code '1'`

Is this line right?
=> SRC_URI += "file://hooks.sh"
=> or do I have to write: SRC_URI += "${THISDIR}/files/hook.sh"

Test4564, 2021-03-24:
I solved the problem with the license.
I try tomorrow if your suggested lines work!

ejoerns, 2021-03-24:
@Test4564 bitbake wants you to have a license when you include files as one can assume that this file is under a certain license. The simples case is to put it under MIT and do

```
LIC_FILES_CHKSUM = "file://${COREBASE}/meta/COPYING.MIT;md5=3da9cfbcb788c80a0384361b4de20420"
```

Since #178 you are also free to chose a different one. But then you need to set `LIC_FILES_CHKSUM` to the appropriate one of ``${COREBASE}/meta/files/common-licenses/`` or ad a custom file.

Test4564, 2021-03-25:
Thank you!
Yes I solved it in this way.

My update-bundle.bb:
`DESCRIPTION = "RAUC bundle generator"

LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/MIT;md5=0835ade698e0bcf8506ecda2f7b4f302"

inherit bundle

RAUC_BUNDLE_COMPATIBLE = "RaspberryPi4"
RAUC_BUNDLE_VERSION = "v20200703"
RAUC_BUNDLE_DESCRIPTION = "RAUC Demo Bundle"
RAUC_BUNDLE_SLOTS = "rootfs" 
RAUC_SLOT_rootfs = "core-image-minimal"
RAUC_SLOT_rootfs[fstype] = "ext4"

RAUC_KEY_FILE = "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"

SRC_URI += " file://hook.sh"
RAUC_BUNDLE_HOOKS[file] = "hook.sh"
RAUC_BUNDLE_HOOKS[hooks] = "post-install"`

My hook.sh:
`#!/bin/sh

case "$1" in
        slot-post-install)
                # only rootfs needs to be handled
                test "$RAUC_SLOT_CLASS" = "rootfs" || exit 0

                touch "$RAUC_SLOT_MOUNT_POINT/extra-file"
                ;;
        *)
                exit 1
                ;;
esac

exit 0`

My generated manifest.raucm
`[update]
compatible=RaspberryPi4
version=v20200703
description=RAUC Demo Bundle
build=20210325063351

[hooks]
filename=hook.sh

[image.rootfs]
sha256=aeb8fa180acb027d62ea566749956b5809f6ff850c3f8f1539e5ce46d666190d
size=176160768
filename=core-image-minimal-raspberrypi3.ext4`

If I update my raspberry with this bundle.
I can not see any effect of the hook.sh...
I would have expected that I can see after the update a file named "extra-file" in the root directory.
What is my failure here?
=> I wonder that in the generated manifest.raucm is nothing to see about the "post-install" ...?

Extract from the documentation:
[image]

Test4564, 2021-04-01:
Do you know whats wrong with my configuration?
Why is my post-install hook not working?

Do I misunderstand the example out of the documentation?

ejoerns, 2021-04-06:
@Test4564 instead of

```
RAUC_BUNDLE_HOOKS[hooks] = "post-install"
```

which is for speciying 'Install Hooks' [[1]](https://rauc.readthedocs.io/en/latest/using.html#install-hooks),
you want to specify a 'Slot Hook' [[2]](https://rauc.readthedocs.io/en/latest/using.html#slot-hooks):

```
RAUC_SLOT_rootfs[hooks] = "post-install"
```

You should at least see any warning anywhere when doing this wrong. Don't you?

alexanderwwagner, 2021-09-07:
Yes I have one boot partition (kernel + bootloader) and one rootfs partition.
(Thats the default configuration of yocto for this system.)

Okay I am going to research about this topic.
Maybe I change it to your suggestion and put the kernel in the rootfs but  I have to get some informations about this.
At the moment I do not have any idea what I have to do, that yocto is doing this.

Thank you!

ejoerns, 2021-09-24:
> Okay I am going to research about this topic.
> Maybe I change it to your suggestion and put the kernel in the rootfs but I have to get some informations about this.

@alexanderwwagner It should be as easy as putting `kernel-image` and `kernel-devicetree` into IMAGE_INSTALL

What we then additionally use here (but this needs to be supported by the bootloader) is the `Bootloader Specification` for booting the right combination of kernel + dtb (see https://systemd.io/BOOT_LOADER_SPECIFICATION/) with the spec file created using [this](https://github.com/pengutronix/meta-ptx/blob/master/classes/bootspec.bbclass) class. This needs to be supported by the bootloader (as it is in barebox), but you can also simply tell the bootloder explicitly which kernel + dtb to boot from the rootfs.

closed 2024-07-03 by ejoerns