rauc/meta-rauc #210: Using meta-rauc for Intel 64 bit machine with UEFI secure boot

qorinator, 2021-12-01:
Hi everyone,

I am currently building a yocto genericx86-64 machine for an Intel 64 bit machine with UEFI secure boot and would like to use meta-rauc as the system update mechanism. 

However I encountered a few issues.
1. I am not able to see .raucb bundle deployed in the deploy dir and I also don't see the do_bundle is called anywhere in the log.
2. rauc gives off this warning in the boot log
```
(rauc:468) : rauc-WARNING **: FAILED getting primary slot: Failed getting primary slot: Did not find primary boot entry!
rauc-message: rauc status: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.code30: Failed marking 'rootfs.0' as good: No entry for bootname 'A' found.
```
3. rauc status doesn't show the activated bootloader and the boot status is marked as bad
[image]

Here is the update-bundle.bb
```
inherit bundle

RAUC_BUNDLE_COMPATIBLE = "generic86-64"
RAUC_BUNDLE_VERSION = "v2021-12-01"

RAUC_BUNDLE_SLOTS = "efi rootfs"

RAUC_SLOT_rootfs = "ipc-core-image"

RAUC_SLOT_efi = "bootimg-efi"
RAUC_SLOT_efi[type] = "boot"

RAUC_IMAGE_FSTYPE = "ext4"

RAUC_KEY_FILE = "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE = "${THISDIR}/files/development-1.cert.pem"
```
 
the system.conf
```
[system]
compatible=generic86-64
bootloader=efi
statusfile=/data/central-status.raucs
bundle-formats=-plain

[slot.efi.0]
device=/dev/sda1

[slot.rootfs.0]
device = /dev/sda2
type = ext4
bootname=A

[slot.rootfs.1]
device = /dev/sda3
type = ext4
bootname=B
```

I am new at yocto and at RAUC. would be nice if someone can help to point at which direction I need to look at. 

Thank you!!

twinning92, 2021-12-23:
What image are you telling BitBake to build? You need to explicitly call `bitbake update-bundle` on the bundle recipe. I'm having to guess here, but everything here looks pretty good to me. If I had to hazard a guess I'd think you might be trying to add the update-bundle recipe to your image?

qorinator, 2021-12-24:
Hi,

 yes,

You are correct. I missed the part in the readme that said to change the building of the recipe to the update bundle recipe 🤦🏻‍♂️. 

And after that i forgot to close this issue.

closed 2021-12-24 by qorinator

bencorrado, 2024-08-27:
@qorinator Would you mind sharing the rest of what you did to enable x86 secure boot with rauc?