rauc/meta-rauc #220: rauc-hawkbit do_install task failing

MitchGaines, 2022-04-18:
Hi there,

I have noticed that the `do_install` task on the rauc-hawkbit recipe has started failing during my builds when it previously built without issue. I have tried on the dunfell and master branches both with the same issue. It doesn't seem like there has been any meaningful change in the recipe itself, but I am wondering if an upstream change to setuptools has been changed and introduced this failure. 

I cannot say I am too familiar with setuptools. So, it is hard for me to say where it is attempting to get the version information from. Any guidance would be helpful. 

See the log of the error below:

```
mitch:build$ bitbake rauc-hawkbit
Loading cache: 100% |##########################################################| Time: 0:00:00
Loaded 3680 entries from dependency cache.
Parsing recipes: 100% |########################################################| Time: 0:00:03
Parsing of 2446 .bb files complete (2441 cached, 5 parsed). 3687 targets, 137 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "rpi-cm4"
DISTRO               = "yoctoOS"
DISTRO_VERSION       = "22.04"
TUNE_FEATURES        = "aarch64 cortexa72 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:ce50594d709f473fbad30e187565531af6714aaa"
meta-oe              
meta-perl            
meta-python          
meta-filesystems     
meta-networking      = "dunfell:86b864a4d8c28185a4a464583fb86f73aa22847a"
meta-security        = "dunfell:b76698c788cb8ca632077a972031899ef15025d6"
meta-selinux         = "dunfell:1fba53261bb8d7d3a5dcf238e388811cbb15219b"
meta-rauc            = "dunfell:2fe17af894b82cd2b0d0203478d4924fa916b440"
meta-rauc-raspberrypi = "master:f09b79a72ad53f7abbab812a6f287e4b3f447b1a"
meta-clang           = "dunfell:3bb001d3f364bbf6588fed04b9ee2c7e74a0beba"
meta-java            = "dunfell:3af889e112ddb5c677990535b1e0209c08f68e3f"
meta-rust            = "master:c654f5cb928bd4f4c7da7d74a8356fd2a94283f6"
meta-raspberrypi     = "dunfell:934064a01903b2ba9a82be93b3f0efdb4543a0e8"

Initialising tasks: 100% |#####################################################| Time: 0:00:02
Sstate summary: Wanted 15 Found 8 Missed 7 Current 874 (53% match, 99% complete)
NOTE: Executing Tasks
ERROR: rauc-hawkbit-0.2.0+gitAUTOINC+47bebb4a01-r0 do_install: 'python3 setup.py install --root=/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.8/site-packages     --install-data=/usr/share' execution failed.
ERROR: rauc-hawkbit-0.2.0+gitAUTOINC+47bebb4a01-r0 do_install: Execution of '/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/temp/run.do_install.234086' failed with exit code 1
ERROR: Logfile of failure stored in: /path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/temp/log.do_install.234086
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/path/to/yoctoOS/layers/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/path/to/yoctoOS/layers/poky/meta/recipes-core/glibc/glibc_2.31.bb:do_populate_sysroot', 'virtual:native:/path/to/yoctoOS/layers/poky/meta/recipes-devtools/python/python3_3.8.13.bb:do_populate_sysroot', 'virtual:native:/path/to/yoctoOS/layers/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/path/to/yoctoOS/layers/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/path/to/yoctoOS/layers/poky/meta/recipes-devtools/python/python3-setuptools_45.2.0.bb:do_populate_sysroot', '/path/to/yoctoOS/layers/poky/meta/recipes-devtools/gcc/gcc-cross_9.3.bb:do_populate_sysroot', '/path/to/yoctoOS/layers/poky/meta/recipes-devtools/python/python3_3.8.13.bb:do_populate_sysroot', 'virtual:native:/path/to/yoctoOS/layers/third-party/meta-openembedded/meta-python/recipes-devtools/python/python3-setuptools-scm_3.3.3.bb:do_populate_sysroot', '/path/to/yoctoOS/layers/poky/meta/recipes-devtools/gcc/gcc-runtime_9.3.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['quilt-native', 'glibc', 'python3-native', 'pseudo-native', 'patch-native', 'python3-setuptools-native', 'gcc-cross-aarch64', 'python3', 'python3-setuptools-scm-native', 'gcc-runtime', 'linux-libc-headers', 'zlib-native', 'sqlite3-native', 'libtirpc-native', 'autoconf-native', 'gnu-config-native', 'xz-native', 'bzip2-native', 'util-linux-native', 'pkgconfig-native', 'libffi-native', 'gdbm-native', 'automake-native', 'readline-native', 'libtool-native', 'openssl-native', 'libnsl2-native', 'attr-native', 'unzip-native', 'mpfr-native', 'libmpc-native', 'binutils-cross-aarch64', 'gmp-native', 'texinfo-dummy-native', 'flex-native', 'libxcrypt', 'zlib', 'opkg-utils', 'autoconf-archive', 'libtirpc', 'readline', 'libffi', 'libnsl2', 'util-linux', 'bzip2', 'openssl', 'xz', 'sqlite3', 'gdbm', 'libgcc', 'm4-native', 'gettext-minimal-native', 'libcap-ng-native', 'libpcre2-native', 'ncurses-native', 'ncurses', 'libcap-ng', 'bash-completion']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| Traceback (most recent call last):
|   File "setup.py", line 6, in <module>
|     setup(
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/__init__.py", line 144, in setup
|     return distutils.core.setup(**attrs)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/distutils/core.py", line 108, in setup
|     _setup_distribution = dist = klass(attrs)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/dist.py", line 447, in __init__
|     _Distribution.__init__(self, {
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/distutils/dist.py", line 292, in __init__
|     self.finalize_options()
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/dist.py", line 740, in finalize_options
|     ep.load()(self)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools/dist.py", line 747, in _finalize_setup_keywords
|     ep.load()(self, ep.name, value)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools_scm/integration.py", line 17, in version_keyword
|     dist.metadata.version = get_version(**value)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 150, in get_version
|     parsed_version = _do_parse(config)
|   File "/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/setuptools_scm/__init__.py", line 105, in _do_parse
|     raise LookupError(
| LookupError: setuptools-scm was unable to detect version for '/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/git'.
| 
| Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
| 
| For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
[... cut ...]
ERROR: Task (/path/to/yoctoOS/layers/third-party/meta-rauc/recipes-support/rauc-hawkbit/rauc-hawkbit_git.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2737 tasks of which 2728 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /path/to/yoctoOS/layers/third-party/meta-rauc/recipes-support/rauc-hawkbit/rauc-hawkbit_git.bb:do_install
Summary: There were 4 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

MitchGaines, 2022-04-22:
So, to add onto this issue a little bit, I was able to replicate the issue outside of the yocto environment by running `/path/to/sysroot/python3 ../git/setup.py install --root=/path/to/yoctoOS/build/tmp/work/aarch64-poky-linux/rauc-hawkbit/0.2.0+gitAUTOINC+47bebb4a01-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.8/site-packages     --install-data=/usr/share`. To be clear though, I am not able to replicate the issue when I run this from the `/git` directory

I found [this issue](https://github.com/pypa/pip/issues/1979) with setuptools related to the directory it was being run from. So, I added the following patch to the `setup.py`

```
diff --git a/setup.py b/setup.py
index 92cacb3..abdd570 100755
--- a/setup.py
+++ b/setup.py
@@ -1,8 +1,11 @@
 #!/usr/bin/env python
 # -*- coding: utf-8 -*-
 
+import os
 from setuptools import setup, find_packages
 
+os.chdir(os.path.abspath(os.path.dirname(__file__)))
+
 setup(
       name='rauc-hawkbit',
       description='hawkBit client for RAUC',
```

Unfortunately, that doesn't seem to have had an impact. Also, as far as I can tell from the do_install task logs, these commands should be getting run from the `/git` directory anyway. When I mimic the yocto environment and run the commands it runs, it works without issue. So, that's odd. 

I also hit this issue on another unrelated project I am working on that uses the same tools for the most part. This makes me think this is not just an issue in my environment. Has anyone been able to replicate this issue?

Bastian-Krause, 2022-05-02:
Note that rauc-hawkbit is not intended for production use. Consider using [rauc-hawkbit-updater](https://github.com/rauc/rauc-hawkbit-updater) instead.

> I have noticed that the `do_install` task on the rauc-hawkbit recipe has started failing during my builds when it previously built without issue. I have tried on the dunfell and master branches both [...]

So, which change broke the `do_install` task then?

In our GitHub actions CI, [rauc-hawkbit on dunfell builds fine](https://github.com/rauc/meta-rauc/runs/5069467218).

MitchGaines, 2022-05-26:
@Bastian-Krause I just removed `rauc-hawkbit` from my build as I realized it wasn't necessary for what I need. I know its unhelpful, but I just wasn't able to resolve this issue.

surabhiashok1, 2022-05-27:
I want to use rauc-hawkbit client in the yocto project. I want to start the hawkbit client on board and communicate with the eclipse hawkbit server. So, I need rauc-hawkbit support for the meta-rauc thud version layer. I am facing the following issue. Can anyone please help?

ERROR: rauc-hawkbit-0.1.0+gitAUTOINC+79421c1c0b-r0 do_install: 'python3 setup.py install --root=apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.5/site-packages     --install-data=/usr/share' execution failed.
ERROR: rauc-hawkbit-0.1.0+gitAUTOINC+79421c1c0b-r0 do_install: Function failed: do_install (log file is located at apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624)
ERROR: Logfile of failure stored in: apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['apps_proc/poky/meta/recipes-devtools/gcc/gcc-cross_8.2.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/python/python3_3.5.6.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/gcc/gcc-runtime_8.2.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta-openembedded/meta-python/recipes-devtools/python/python3-setuptools-scm_2.1.0.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-core/glibc/glibc_2.28.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta/recipes-devtools/python/python3-setuptools_40.0.0.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/python/python3-native_3.5.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gcc-cross-aarch64', 'quilt-native', 'python3', 'gcc-runtime', 'pseudo-native', 'python3-setuptools-scm-native', 'glibc', 'python3-setuptools-native', 'python3-native', 'libtool-native', 'zlib-native', 'flex-native', 'binutils-cross-aarch64', 'libmpc-native', 'gmp-native', 'mpfr-native', 'autoconf-native', 'xz-native', 'automake-native', 'linux-libc-headers', 'texinfo-dummy-native', 'gnu-config-native', 'readline', 'xz', 'gdbm', 'sqlite3', 'libxcrypt', 'bzip2', 'opkg-utils', 'libffi', 'openssl', 'zlib', 'libgcc', 'unzip-native', 'openssl-native', 'readline-native', 'gdbm-native', 'sqlite3-native', 'pkgconfig-native', 'bzip2-native', 'gettext-minimal-native', 'm4-native', 'ncurses', 'ncurses-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| Traceback (most recent call last):
|   File "setup.py", line 23, in <module>
|     'bin/rauc-hawkbit-client'
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/__init__.py", line 131, in setup
|     return distutils.core.setup(**attrs)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/distutils/core.py", line 108, in setup
|     _setup_distribution = dist = klass(attrs)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/dist.py", line 370, in __init__
|     k: v for k, v in attrs.items()
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
|     self.finalize_options()
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/dist.py", line 529, in finalize_options
|     ep.load()(self, ep.name, value)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/integration.py", line 22, in version_keyword
|     dist.metadata.version = get_version(**value)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/__init__.py", line 119, in get_version
|     parsed_version = _do_parse(root, parse)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/__init__.py", line 97, in _do_parse
|     "use git+https://github.com/user/proj.git#egg=proj" % root)
[... cut ...]
| WARNING: apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/run.do_install.8624:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_install (log file is located at apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624)
ERROR: Task (apps_proc/poky/meta-rauc-thud/recipes-support/rauc-hawkbit/rauc-hawkbit_git.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7431 tasks of which 7430 didn't need to be rerun and 1 failed.

Bastian-Krause, 2022-05-27:
[thud is EOL since almost three years](https://wiki.yoctoproject.org/wiki/Releases), so you should not expect a lot of help here. I suspect that it might have something to do with the setuptools-scm version.

Also:

> Note that rauc-hawkbit is not intended for production use. Consider using [rauc-hawkbit-updater](https://github.com/rauc/rauc-hawkbit-updater) instead.

ejoerns, 2022-05-27:
Our CI builds this package without any notable error.

What I discover from both logs (thud and dunfell) is that they have in common to build for **aarch64**.

This we do neither test in CI nor manually. I'll start an aarch64 build locally to see if I also run into this.

ejoerns, 2022-05-27:
Compiles fine on aarch64 as well here with poky.

I ran

```
bitbake rauc-hawkbit 
Parsing recipes: 100% |#######################################################################################| Time: 0:00:19
Parsing of 1916 .bb files complete (0 cached, 1916 parsed). 2987 targets, 111 skipped, 0 masked, 0 errors.
WARNING: No bb files matched BBFILE_PATTERN_yoctobsp '^/ptx/work/WORK_SANAH/ejo/Code/meta-rauc/test-build/poky/meta-yocto-bsp/'
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "qemuarm64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.16"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:232b5533de22bba7a747f73f684d609e217531f9"
meta-oe              
meta-python          = "dunfell:8ff12bfffcf0840d5518788a53d88d708ad3aae0"
meta-rauc            = "dunfell:2fe17af894b82cd2b0d0203478d4924fa916b440"
```

@MitchGaines If you are interested in debugging, this: Another option might be to see if this works or breaks with plain poky on your side. Otherwise, maybe something in your setup (e.g. your distro) has an impact on the build failure.

If debugging this on dunfell is not of interest, I would like to close this. I will not attempt to debug this on thud.

surabhiashok1, 2022-05-28:
> Note that rauc-hawkbit is not intended for production use. Consider using [rauc-hawkbit-updater](https://github.com/rauc/rauc-hawkbit-updater) instead.

Can you please help how to rauc-hawkbit-updater with yocto project.

surabhiashok1, 2022-05-28:
Can you clarify about building rauc and rauc-hawkbit together?

surabhiashok1, 2022-05-28:
**support for python < 3.6 has been removed in setuptools_scm>=6.0.0.** 
In the error log, the python 3.5 is taking. Can you please tell where i can change it to >3.6 python version so that i can check whether it is with the scm lookup error.

> I suspect that it might have something to do with the setuptools-scm version.

Bastian-Krause, 2022-05-28:
Again, Yocto's thud release is EOL. You should not be using this in the first place and you cannot expect support for that.

Our meta-rauc issue tracker is no Yocto help channel. You need to learn some Yocto basics first (hint: native packages), then you might ask on the Yocto mailing lists or their IRC channels.

> > Note that rauc-hawkbit is not intended for production use. Consider using [rauc-hawkbit-updater](https://github.com/rauc/rauc-hawkbit-updater) instead.
> 
> Can you please help how to rauc-hawkbit-updater with yocto project.

I am sure you'll find the appropriate recipe in the _latest_ Yocto release branches of meta-rauc.

surabhiashok1, 2022-05-29:
> I am sure you'll find the appropriate recipe in the _latest_ Yocto release branches of meta-rauc.

Thank you.
Can you please give clarity on this?

**support for python < 3.6 has been removed in setuptools_scm>=6.0.0.
In the error log, the python 3.5 is taking.**

and any more suggestion to resolve the build error of rauc-hawkbit (which i commented previously) would be very helpful.

MitchGaines, 2022-05-30:
@Bastian-Krause @ejoerns Sorry, I just haven't been focused on this issue for the past few weeks as I considered it broken and didn't have the time to fix it. I just pulled the latest poky source (dunfell) and all tasks ran without issue. Must have been something that changed in poky; I'm unsure of what though.

ejoerns, 2022-05-31:
> @Bastian-Krause @ejoerns Sorry, I just haven't been focused on this issue for the past few weeks as I considered it broken and didn't have the time to fix it. I just pulled the latest poky source [...]

@MitchGaines Thanks for the update! :+1: So I guess we can close this

> > I am sure you'll find the appropriate recipe in the _latest_ Yocto release branches of meta-rauc.
> 
> Thank you. Can you please give clarity on this?
> 
> **support for python < 3.6 has been removed in setuptools_scm>=6.0.0. In the error log, the python 3.5 is taking.**

@surabhiashok1 I do not understand how you come to the conclusion that this leads to the noted error. Could be...
As this is a python-specific question, please ask in python support channels.

> and any more suggestion to resolve the build error of rauc-hawkbit (which i commented previously) would be very helpful.

It is your decision and your right to use stuff from the past, but please do not expect support for this.
With `dunfell` we have an LTS branch in Yocto that is explicitly supported for ~4 years.
This also comes with more recent python. Just update all your layers at least to dunfell.

If anyone happens to know the solution, feel free to comment but I will close this now.

closed 2022-05-31 by ejoerns