rauc/meta-rauc #222: rauc-hawkbit_git.bb:do_install failed with exit code '1'

surabhiashok1, 2022-05-26:
meta-rauc thud version giving the following error while building RAUC hawkBit client.

surabhiashok1, 2022-05-26:
More details about the error:

ERROR: rauc-hawkbit-0.1.0+gitAUTOINC+79421c1c0b-r0 do_install: 'python3 setup.py install --root=apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.5/site-packages     --install-data=/usr/share' execution failed.
ERROR: rauc-hawkbit-0.1.0+gitAUTOINC+79421c1c0b-r0 do_install: Function failed: do_install (log file is located at apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624)
ERROR: Logfile of failure stored in: apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['apps_proc/poky/meta/recipes-devtools/gcc/gcc-cross_8.2.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/python/python3_3.5.6.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/gcc/gcc-runtime_8.2.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta-openembedded/meta-python/recipes-devtools/python/python3-setuptools-scm_2.1.0.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-core/glibc/glibc_2.28.bb:do_populate_sysroot', 'virtual:native:apps_proc/poky/meta/recipes-devtools/python/python3-setuptools_40.0.0.bb:do_populate_sysroot', 'apps_proc/poky/meta/recipes-devtools/python/python3-native_3.5.6.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['gcc-cross-aarch64', 'quilt-native', 'python3', 'gcc-runtime', 'pseudo-native', 'python3-setuptools-scm-native', 'glibc', 'python3-setuptools-native', 'python3-native', 'libtool-native', 'zlib-native', 'flex-native', 'binutils-cross-aarch64', 'libmpc-native', 'gmp-native', 'mpfr-native', 'autoconf-native', 'xz-native', 'automake-native', 'linux-libc-headers', 'texinfo-dummy-native', 'gnu-config-native', 'readline', 'xz', 'gdbm', 'sqlite3', 'libxcrypt', 'bzip2', 'opkg-utils', 'libffi', 'openssl', 'zlib', 'libgcc', 'unzip-native', 'openssl-native', 'readline-native', 'gdbm-native', 'sqlite3-native', 'pkgconfig-native', 'bzip2-native', 'gettext-minimal-native', 'm4-native', 'ncurses', 'ncurses-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| Traceback (most recent call last):
|   File "setup.py", line 23, in <module>
|     'bin/rauc-hawkbit-client'
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/__init__.py", line 131, in setup
|     return distutils.core.setup(**attrs)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/distutils/core.py", line 108, in setup
|     _setup_distribution = dist = klass(attrs)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/dist.py", line 370, in __init__
|     k: v for k, v in attrs.items()
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/distutils/dist.py", line 281, in __init__
|     self.finalize_options()
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools/dist.py", line 529, in finalize_options
|     ep.load()(self, ep.name, value)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/integration.py", line 22, in version_keyword
|     dist.metadata.version = get_version(**value)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/__init__.py", line 119, in get_version
|     parsed_version = _do_parse(root, parse)
|   File "apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/recipe-sysroot-native/usr/lib/python3.5/site-packages/setuptools_scm/__init__.py", line 97, in _do_parse
|     "use git+https://github.com/user/proj.git#egg=proj" % root)
| LookupError: setuptools-scm was unable to detect version for 'apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/git'.
| 
[... cut ...] -lib=/usr/lib/python3.5/site-packages     --install-data=/usr/share' execution failed.
| WARNING: apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/run.do_install.8624:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_install (log file is located at apps_proc/build-qti-distro-fullstack-virtualization-debug/tmp-glibc/work/aarch64-oe-linux/rauc-hawkbit/0.1.0+gitAUTOINC+79421c1c0b-r0/temp/log.do_install.8624)

Bastian-Krause, 2022-05-26:
Duplicate of #220. Please continue the discussion there.