rauc/meta-rauc #303: ISSUE: post-install hook after uboot bundle.

umarFarooq-emb, 2024-02-14:
I am trying to update my uboot image using RAUC,, this slot contains a bootloader(SSBL) and this partition is not ext4,, I want to run a post-install script only when my ssbl(uboot) is updated. To do this I am using a post install hook like follows in the bundle

```py
RAUC_BUNDLE_HOOKS[file] = "update-metadata.sh"
RAUC_SLOT_fip-senec-os[hooks] = "pre-install"
```

and I am getting the following logs while the bundle is updated in the system

```
 60% Checking manifest contents done.
 60% Determining target install group
 80% Determining target install group done.
 80% Updating slots
[13910.481015] EXT4-fs (mmcblk1p6): VFS: Can't find ext4 filesystem
 80% Checking slot fip-senec-os.1
 90% Checking slot fip-senec-os.1 done.
 90% Copying image to fip-senec-os.1
 99% Copying image to fip-senec-os.1 failed.
 99% Updating slots failed.
100% Installing failed.
LastError: Installation error: Failed updating slot fip-senec-os.1: failed to mount slot: failed to run mount: Child process exited with code 32
```

If i can use handlers? if using the handlers I can run a script only when my uboot is updated? or If using some flag I can stop the post-install hook from mounting uboot partition?

~Umar

ejoerns, 2024-02-14:
@umarFarooq-emb Maybe you have erroneously configured your non-ext4 slot as `type=ext4`?
If that's the case, please try to replace it with `type=raw`.

For a mountable file system, like ext4, it should be a safe assumption to be able to mount the slot after the update.
This is documented in the 'post-install' section of https://rauc.readthedocs.io/en/latest/using.html#slot-hooks

closed 2024-02-14 by umarFarooq-emb

jluebbe, 2024-02-14:
> I am trying to update my uboot image using RAUC,, this slot contains a bootloader(SSBL) and this partition is not ext4,, I want to run a post-install script only when my ssbl(uboot) is updated. To do this I am using a post install hook like follows in the bundle
> 
> ```python
> RAUC_BUNDLE_HOOKS[file] = "update-metadata.sh"
> RAUC_SLOT_fip-senec-os[hooks] = "pre-install"
> ```

You say post install hook, but specify `pre-install` in the config?

umarFarooq-emb, 2024-02-14:
@jluebbe for the sake of testing I tried both the post-install and pre-install hooks but it always returned the same ext4 slot mount error.

The issue was in the system.conf file, where I changed the slot `type=ext4` to `type=raw`. 

It is resolved now. Thanks