rauc/meta-rauc #315: task do_deploy_archives: Failed

zonque, 2024-04-30:
When Yocto builds are interrupted, we frequently see the following error.
That seems to be a recent regression.

```
10:23:26  ERROR: update-bundle-1.0-r0 do_deploy_archives: Error executing a python function in exec_func_python() autogenerated:
10:23:26  
10:23:26  The stack trace of python calls that resulted in this exception/failure was:
10:23:26  File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
10:23:26       0001:
10:23:26   *** 0002:sstate_task_postfunc(d)
10:23:26       0003:
10:23:26  File: '/var/sstate/latest/build/machine/poky/build/conf/../../../poky/meta/classes-global/sstate.bbclass', lineno: 801, function: sstate_task_postfunc
10:23:26       0797:
10:23:26       0798:    omask = os.umask(0o002)
10:23:26       0799:    if omask != 0o002:
10:23:26       0800:       bb.note("Using umask 0o002 (not %0o) for sstate packaging" % omask)
10:23:26   *** 0801:    sstate_package(shared_state, d)
10:23:26       0802:    os.umask(omask)
10:23:26       0803:
10:23:26       0804:    sstateinst = d.getVar("SSTATE_INSTDIR")
10:23:26       0805:    d.setVar('SSTATE_FIXMEDIR', shared_state['fixmedir'])
10:23:26  File: '/var/sstate/latest/build/machine/poky/build/conf/../../../poky/meta/classes-global/sstate.bbclass', lineno: 714, function: sstate_package
10:23:26       0710:    for f in (d.getVar('SSTATECREATEFUNCS') or '').split() + \
10:23:26       0711:             sstate_create_package + \
10:23:26       0712:             (d.getVar('SSTATEPOSTCREATEFUNCS') or '').split():
10:23:26       0713:        # All hooks should run in SSTATE_BUILDDIR.
10:23:26   *** 0714:        bb.build.exec_func(f, d, (sstatebuild,))
10:23:26       0715:
10:23:26       0716:    # SSTATE_PKG may have been changed by sstate_report_unihash
10:23:26       0717:    siginfo = d.getVar('SSTATE_PKG') + ".siginfo"
10:23:26       0718:    if not os.path.exists(siginfo):
10:23:26  File: '/var/sstate/latest/build/machine/poky/bitbake/lib/bb/build.py', lineno: 260, function: exec_func
10:23:26       0256:    with bb.utils.fileslocked(lockfiles):
10:23:26       0257:        if ispython:
10:23:26       0258:            exec_func_python(func, d, runfile, cwd=adir)
10:23:26       0259:        else:
10:23:26   *** 0260:            exec_func_shell(func, d, runfile, cwd=adir)
10:23:26       0261:
10:23:26       0262:    try:
10:23:26       0263:        curcwd = os.getcwd()
10:23:26       0264:    except:
10:23:26  File: '/var/sstate/latest/build/machine/poky/bitbake/lib/bb/build.py', lineno: 585, function: exec_func_shell
10:23:26       0581:                better_frames = ("\t{0}".format(frame) for frame in better_frames)
10:23:26       0582:                exe.extra_message = "\nBacktrace (metadata-relative locations):\n{0}".format("\n".join(better_frames))
10:23:26       0583:            raise
10:23:26       0584:        finally:
10:23:26   *** 0585:            os.unlink(fifopath)
10:23:26       0586:
10:23:26       0587:    bb.debug(2, "Shell function %s finished" % func)
10:23:26       0588:
10:23:26       0589:def _task_data(fn, task, d):
10:23:26  Exception: FileNotFoundError: [Errno 2] No such file or directory: '/var/sstate/latest/build/machine/poky/build/tmp-glibc/work/machine-linux-gnueabi/update-bundle-embedded/1.0/temp/fifo.276626'
10:23:26  
10:23:26  NOTE: recipe update-bundle-embedded-1.0-r0: task do_deploy_archives: Failed
```

ejoerns, 2024-04-30:
@zonque This looks like the archiver class is used, correct?

For which poky/bitbake version do you see this issue?

zonque, 2024-04-30:
We"re on `nanbield` right now.