rauc/meta-rauc #321: Multiple hooks targets

gbetous, 2024-06-05:
Hi,

I can't find the syntax for a Yocto recipe for having the hook script triggered for multiple targets.

I tried:

`RAUC_SLOT_rootfs[hooks] = "install post-install"`

`RAUC_SLOT_rootfs[hooks] = "install; post-install"`

and even

```
RAUC_SLOT_rootfs[hooks] = "install"
RAUC_SLOT_rootfs[hooks] += "post-install"
```

I couldn't find any example in the documentation nor generally Googling.

Thanks !

ejoerns, 2024-06-06:
The header in [bundle.bbclass](https://github.com/rauc/meta-rauc/blob/master/classes-recipe/bundle.bbclass#L19) says

    RAUC_SLOT_rootfs[hooks] ?= "install;post-install"

This doesn't work for you?

gbetous, 2024-06-06:
Thanks for your reply.

With this syntax I have a different behavior : there is no syntax error, but post-install hook is not triggered.

With `rauc info` I can see that the hooks are present is my generated package : `Hooks:     install post-install`

But this being said, when I install it, the post-install hook is not triggered:

```
juin 05 19:04:18 product rauc[310]: Active slot bootname: rootB
juin 05 19:04:18 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: started
juin 05 19:04:18 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Checking and mounting bundle...
juin 05 19:04:18 product rauc[310]: Reading bundle: /home/root/rauc-bundle-product.raucb
juin 05 19:04:18 product rauc[310]: Verifying bundle signature... 
juin 05 19:04:18 product rauc[310]: Verified inline signature by 'O = Test Org, CN = Test Org Development-1'
juin 05 19:04:18 product rauc[310]: Mounting bundle '/home/root/rauc-bundle-product.raucb' to '/run/rauc/bundle'
juin 05 19:04:18 product rauc[310]: Configured loop device '/dev/loop0' for 61845504 bytes
juin 05 19:04:18 product rauc[310]: Configured dm-verity device '/dev/dm-0'
juin 05 19:04:18 product rauc[310]: Marking target slot rootfs.0 as non-bootable...
juin 05 19:04:20 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Updating slots...
juin 05 19:04:20 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Checking slot rootfs.0
juin 05 19:04:20 product rauc[310]: Updating /dev/mmcblk0p2 with 'install' slot hook
juin 05 19:04:20 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Updating slot rootfs.0
juin 05 19:04:20 product rauc[310]: Running slot hook 'slot-install' for rootfs.0
juin 05 19:04:20 product rauc[482]: SLOT INSTALL
juin 05 19:04:20 product rauc[482]: RAUC_IMAGE_NAME: /run/rauc/bundle/production-image-product.ext4.verity
juin 05 19:04:20 product rauc[482]: RAUC_SLOT_DEVICE: /dev/mmcblk0p2
juin 05 19:04:20 product rauc[310]: Marking target slot rootfs.0 as bootable...
juin 05 19:04:20 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Updating slot rootfs.0 status
juin 05 19:04:20 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: Updating slot rootfs.0 done
juin 05 19:04:22 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: All slots updated
juin 05 19:04:22 product rauc[310]: installing /home/root/rauc-bundle-product.raucb: finished
juin 05 19:04:22 product rauc[310]: installing `/home/root/rauc-bundle-product.raucb` succeeded
```

ejoerns, 2024-06-06:
Ouh, that's not ideal on several levels (and I didn't notice in the first run either):

When using an `install` hook, `post-install` hook is expected to **not** run. This is documented here:
https://rauc.readthedocs.io/en/latest/using.html#slot-hooks

The reason is that you can already do everything necessary from the `install` hook.

-> The meta-rauc example is actually bad obviously
-> It would probably be useful if RAUC could warn for configurations like yours (during bundle creation)

I leave this PR open as a reference that this should probably be fixed.

closed 2024-10-30 by jluebbe