rauc/meta-rauc #335: bundle: RAUC_META_mysection[metadata] isn't expending variables anymore (kirkstone -> scarthgap)

FederAndInk, 2024-07-24:
After the kirkstone -> scarthgap migration, meta varflags using yocto variables isn't working anymore

e.g:

```bash
RAUC_META_SECTIONS += "mysection"
RAUC_META_mysection[machine-name] = "${DISTRO}_${MACHINE}_${DISTRO_VERSION}"
```

will inject the `mysection: machine-name=${DISTRO}_${MACHINE}_${DISTRO_VERSION}` into the manifest as is

FederAndInk, 2024-07-24:
It looks like it's been introduced here:
https://github.com/rauc/meta-rauc/commit/ffba61f2f79ec03395629d8713a59226b0158d5f#diff-e5c2ad51d1ccb7561b62ea8e03cc127183911bb93f3b109efb41479a5d0d2631L334-L339

When changing `d.getVarFlag` to `d.getVarFlags`, `d.getVarFlag` expand parameter defaults to `True`, while `d.getVarFlags` does not.

jluebbe, 2024-07-24:
For `d.getVarFlags`, one needs to pass a list of flags to expand, not just true.

It seems that the change to the metadata handling should be reverted. Do you want to make a PR?

FederAndInk, 2024-07-24:
Sure

commit f4d80f95 (2024-07-25): bundle.bbclass: fix variable expansion of RAUC_META_* varflags

Variables assigned to `RAUC_META_<section>[varflag]` are not expanded
anymore. This is caused by `getVarFlags` not expanding variables by
default, revert to using `getVarFlag` which does expand variables by
default.

Fixes: #335
Fixes: ffba61f2f79e ("bundle.bbclass: simplify varflag usage")

(cherry picked from commit 792b1cc318fba6e3267a68674353b1fb015ebd35)
PR #336 (2024-07-25): bundle.bbclass: fix variable expansion of RAUC_META_* varflags
variables assigned to `RAUC_META_<section>[varflag]` are not expanded anymore.

This is caused by `getVarFlags` not expanding variables by default, revert to using `getVarFlag` which does expand variables by default.

fix #335

partially revert changes introduced in ffba61f2f79ec03395629d8713a59226b0158d5f
commit 8fd3907b (2024-07-25): bundle.bbclass: fix variable expansion of RAUC_META_* varflags

Variables assigned to `RAUC_META_<section>[varflag]` are not expanded
anymore. This is caused by `getVarFlags` not expanding variables by
default, revert to using `getVarFlag` which does expand variables by
default.

Fixes: #335
Fixes: ffba61f2f79e ("bundle.bbclass: simplify varflag usage")
commit dcbb8715 (2024-09-09): bundle.bbclass: fix variable expansion of RAUC_META_* varflags

Variables assigned to `RAUC_META_<section>[varflag]` are not expanded
anymore. This is caused by `getVarFlags` not expanding variables by
default, revert to using `getVarFlag` which does expand variables by
default.

Fixes: #335
Fixes: ffba61f2f79e ("bundle.bbclass: simplify varflag usage")

(cherry picked from commit 8fd3907bf5658004d7e6ff2ecdd1efa9f26030e7)
closed 2024-07-25 by jluebbe