rauc/meta-rauc #339: Issue updating from gatesgarth to scarthgap

tr3ysmith, 2024-08-08:
Hey everyone,

So I've been working on a new update for my deployments and the original image was done using Yocto gatesgarth, now in this new update I'm using scarthgap, and when I build the bundle and install it, it all installs fine, but then when I try to boot to that slot, I get an error from Grub saying "Unknown Filesystem". I know that the grub version on the old images is 2.04 and scarthgap grub is 2.12. Could this be a grub issue or a rauc issue? I checked the integrity of the filesystem on the newly updated rootfs slot and it checked out fine. I can also do updates between scarthgap images, just not from gatesgarth to scarthgap... I've looked through all my Rauc configs and everything seems to be the same. Any ideas?

tr3ysmith, 2024-08-08:
I should also note, the new rootfs in the bundle is using kernel 6.6.30, whereas the old rootfs was 5.10. But I since that ships with the rootfs, that shouldn't be a problem right?

tr3ysmith, 2024-08-14:
This was an issue with a new GRUB version, nothing related to rauc

closed 2024-08-14 by tr3ysmith

jluebbe, 2024-08-15:
> This was an issue with a new GRUB version, nothing related to rauc

Thanks for the update! Do you have an issue or discussion you could link to?

tr3ysmith, 2024-08-15:
> > This was an issue with a new GRUB version, nothing related to rauc
> 
> Thanks for the update! Do you have an issue or discussion you could link to?

No, what I found was that grub 2.04 doesn't support booting images built by Yocto scarthgate, maybe something to do with the filesystem, but updating grub to 2.12 seemed to do the trick. I wrote a shell script that is executed in the post-install hook of rauc that updates the bootloader and that did the trick