rauc/meta-rauc #340: Failed mounting bundle: Failed to load dm table: Invalid argument

BhupiMAD, 2024-08-09:
## meta-rauc on custom imx8mp board (Kirkstone).
I want to make RAUC work with a custom imx8mp board. I followed the guide and was able to add `meta-rauc` to my build.
While creating an update bundle and trying to install it on the device I get the below error.

```
root@imx8mp-icore:/# rauc install /home/root/update-bundle-imx8mp-icore-20240809054944.raucb
installing
  0% Installing
  0% Determining slot states
 10% Determining slot states done.
 10% Checking bundle
 10% Verifying signature
 20% Verifying signature done.
 20% Checking bundle done.
 20% Checking manifest contents
 30% Checking manifest contents done.
100% Installing failed.
LastError: Failed mounting bundle: Failed to load dm table: Invalid argument, check DM_VERITY, DM_CRYPT or CRYPTO_AES kernel options.
Installing `/home/root/update-bundle-imx8mp-icore-20240809054944.raucb` failed

```
## rauc info on package
```
root@imx8mp-icore:/# rauc info /home/root/update-bundle-imx8mp-icore-20240809054944.raucb
rauc-Message: 06:23:07.686: Using central status file /data/central.raucs
rauc-Message: 06:23:07.692: valid /etc/rauc/system.conf found, using it
rauc-Message: 06:23:07.692: Reading bundle: /home/root/update-bundle-imx8mp-icore-20240809054944.raucb
rauc-Message: 06:23:07.700: Verifying bundle signature...
rauc-Message: 06:23:07.704: Verified inline signature by 'O = Test Org, CN = Test Org Development-1'
Compatible:     'imx8mp-icore'
Version:        'v20240627'
Description:    'RAUC demonstration bundle'
Build:          '20240809054944'
Hooks:          ''
Bundle Format:  verity
  Verity Salt:  '56fb70e7ed77d83dc646258550b5a984f041001b8c4824b3aaa5d677d75929e1'
  Verity Hash:  'b6e120dee64e99ce80658d0fa758b2610f4f5562b7dd8df654cdd798669ed805'
  Verity Size:  7196672
Manifest Hash:  '92671b6e7f12b5e437162a2b0a09b42b684fd5406e2c850cf3ad2b0b47bea681'

1 Image:
  [rootfs]
        Filename:  dmg-matisse-r5-imx8p-imx8mp-icore.ext4
        Checksum:  3cd96cd3e3f6074370c182314c41cffa207eac9e6338f30da9b58ced6cc4d0ae
        Size:      3.3?GB (3261339648 bytes)
        Hooks:

Certificate Chain:
 0 Subject: O = Test Org, CN = Test Org Development-1
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: 0D:BE:FA:08:AC:C0:01:DF:BE:26:8C:9D:75:DC:A7:A4:1C:8F:10:64:22:1A:9F:04:64:B1:71:00:42:3C:6E:BC
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT
 1 Subject: O = Test Org, CN = Test Org rauc CA Development
   Issuer: O = Test Org, CN = Test Org rauc CA Development
   SPKI sha256: BB:0A:AC:73:FA:86:01:E5:1B:36:24:D3:0D:38:E4:F4:E0:1A:FF:46:8B:BD:20:2C:06:B1:BE:8B:42:92:57:99
   Not Before: Jan  1 00:00:00 1970 GMT
   Not After:  Dec 31 23:59:59 9999 GMT

```
## rauc status
```
root@imx8mp-icore:/# rauc status
=== System Info ===
Compatible:  @@MACHINE@@
Variant:
Booted from: rootfs.0 (A)

=== Bootloader ===
Activated: rootfs.0 (A)

=== Slot States ===
o [rootfs.1] (/dev/mmcblk1p3, ext4, inactive)
        bootname: B
        boot status: good

x [rootfs.0] (/dev/mmcblk1p2, ext4, booted)
        bootname: A
        mounted: /
        boot status: good

```

```
root@imx8mp-icore:/# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda            8:0    1   508M  0 disk /mnt/usb
mmcblk2      179:0    0   7.3G  0 disk
|-mmcblk2p1  179:1    0  83.2M  0 part
`-mmcblk2p2  179:2    0 904.8M  0 part
mmcblk2boot0 179:32   0     4M  1 disk
mmcblk2boot1 179:64   0     4M  1 disk
mmcblk1      179:96   0  59.4G  0 disk
|-mmcblk1p1  179:97   0  83.2M  0 part
|-mmcblk1p2  179:98   0   3.9G  0 part /
|-mmcblk1p3  179:99   0   3.9G  0 part
`-mmcblk1p4  179:100  0  51.4G  0 part /data
```

## update-bundle.bb
```
DESCRIPTION = "RAUC bundle generator"

inherit bundle

RAUC_BUNDLE_FORMAT = "verity"

RAUC_BUNDLE_COMPATIBLE = "${MACHINE}"
RAUC_BUNDLE_VERSION = "v20240627"
RAUC_BUNDLE_DESCRIPTION = "RAUC demonstration bundle"

RAUC_BUNDLE_SLOTS = "rootfs"
RAUC_SLOT_rootfs = "dmg-matisse-r5-imx8p"
RAUC_SLOT_rootfs[fstype] = "ext4"

RAUC_KEY_FILE ?= "${THISDIR}/files/development-1.key.pem"
RAUC_CERT_FILE ?= "${THISDIR}/files/development-1.cert.pem"
```
Should my `RAUC_SLOT_rootfs` be the same as the image originally built for the device? And it makes me wonder why the bundle update files are so huge? the .raucb file that update bundle generates is 800-900 mb.

jluebbe, 2024-08-09:
> LastError: Failed mounting bundle: Failed to load dm table: Invalid argument, check DM_VERITY, DM_CRYPT or CRYPTO_AES kernel options.

Do you have enabled those options in your kernel config? Check `/proc/config.gz` or the config in you build system. Can you attach your full config?

> 1 Image:
>   [rootfs]
>         Size:      3.3?GB (3261339648 bytes)

Your image is 3.3GB, which for ext4 may include empty space.

> |-mmcblk1p2  179:98   0   3.9G  0 part /
> |-mmcblk1p3  179:99   0   3.9G  0 part

> Should my `RAUC_SLOT_rootfs` be the same as the image originally built for the device?

Usually the same rootfs image is used for both the RAUC bundle and the factory disk image.

> And it makes me wonder why the bundle update files are so huge? the .raucb file that update bundle generates is 800-900 mb.

3.3 GB to ~800 MB isn't a bad compression ratio. Which size would you expect?

BhupiMAD, 2024-08-09:
> /proc/config.gz

[config.txt](https://github.com/user-attachments/files/16557673/config.txt)
Here is the complete config file
I found the following
`CONFIG_DM_VERITY=y` is not set
`CONFIG_BLK_DEV_DM=y` is set to `CONFIG_BLK_DEV_DM=m`
`CONFIG_BLK_DEV_NBD=y` is set to `CONFIG_BLK_DEV_NBD=m`

This makes me wonder if my bbappend file is working.

jluebbe, 2024-08-09:
> > /proc/config.gz
> 
> [config.txt](https://github.com/user-attachments/files/16557673/config.txt) Here is the complete config file I found the following `CONFIG_DM_VERITY=y` is not set

That would explain the `Failed to load dm table` error. Also check the other options mentioned in https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration

BhupiMAD, 2024-08-09:
> That would explain the `Failed to load dm table` error. Also check the other options mentioned in https://rauc.readthedocs.io/en/latest/integration.html#kernel-configuration

There are other kernel options that the error showed `DM_CRYPT or CRYPTO_AES` and below are my observations
`# CONFIG_DM_CRYPT is not set`
```
CONFIG_CRYPTO_AES_ARM64=y
CONFIG_CRYPTO_AES_ARM64_CE=y
CONFIG_CRYPTO_AES_ARM64_CE_CCM=y
CONFIG_CRYPTO_AES_ARM64_CE_BLK=y
CONFIG_CRYPTO_AES_ARM64_NEON_BLK=m
CONFIG_CRYPTO_AES_ARM64_BS=m
**CONFIG_CRYPTO_AES=y**
# CONFIG_CRYPTO_AES_TI is not set
```

`CONFIG_CRYPTO_AES` has multiple options, do i need to enable all of them?

jluebbe, 2024-08-09:
As described in the linked documentation `CONFIG_DM_CRYPT` and `CONFIG_CRYPTO_AES` are needed for encrypted bundles only.

BhupiMAD, 2024-08-09:
> 3.3 GB to ~800 MB isn't a bad compression ratio. Which size would you expect?

@jluebbe Thank you for all of your prompt replies.
My question regarding the size was because, for example, I want to add a package like nano editor, will I have to push an 800mb update file for that?

jluebbe, 2024-08-09:
RAUC bundles contain the full system images and are independent of which old version is installed on the system, see https://rauc.readthedocs.io/en/latest/basic.html for more details. Using [adaptive updates](https://rauc.readthedocs.io/en/latest/advanced.html#adaptive-updates) and [streaming](https://rauc.readthedocs.io/en/latest/advanced.html#http-streaming), the download of reused data can be avoided.

BhupiMAD, 2024-08-09:
i got it, my system.conf is the issue. I will update it.

BhupiMAD, 2024-08-09:
@jluebbe I would once again like to **THANK YOU** for all the timely support you provided today.
I was able to test the update and it switches the partition successfully. Will be testing it more. 
My SOC is `Engicam imx8mp`, if there is any way to contribute on what changes are required to bring this board up with RAUC then please let me know and I can do a pull request.

closed 2024-08-09 by BhupiMAD

jluebbe, 2024-08-09:
> My SOC is `Engicam imx8mp`, if there is any way to contribute on what changes are required to bring this board up with RAUC then please let me know and I can do a pull request.

There is https://github.com/rauc/meta-rauc-community. You'd need to check if it could be integrated into an existing layer or should be separate. Alternatively, if you have suggestions to improve the documentation, we'd also appreciate PRs for that: https://github.com/rauc/rauc/tree/master/docs