rauc/meta-rauc #359: Docs link to wrong URL

tropxy, 2024-11-29:
In the Yocto stable docs [here](https://rauc.readthedocs.io/en/stable/integration.html#yocto), the text directs to this link:
 https://github.com/rauc/meta-rauc/blob/master/classes/bundle.bbclass
but it does not exist...Instead, I believe it shall link to this one:
https://github.com/rauc/meta-rauc/blob/master/classes-recipe/bundle.bbclass

ejoerns, 2024-11-30:
@tropxy Indeed. This has been renamed. Thanks for reporting!
Would you like to make a PR against the RAUC documentation to fix it?

ejoerns, 2025-03-04:
Didn't notice this had been fixed by https://github.com/rauc/rauc/pull/1562 already and it actually affected the *stable* version only. Fortunately, with the v1.13 release of RAUC, the fix is in *stable* now.

(For older release versions the link will still be broken, but I don't see a manageable way to resolve this.)

closed 2025-03-04 by ejoerns