rauc/meta-rauc #364: Struggling to use RAUC_BUNDLE_EXTRA_FILES

mortenboye, 2025-01-03:
I've been using rauc successfully for some time, but am now trying to update to the latest.

In the past i've had this line in my recipe
`RAUC_BUNDLE_EXTRA_FILES += "bootfiles"`
because i have a post install script that needs to move some files into place.

However after updating rauc i now get this error:
`Failed to create bundle: directories are not supported as bundle contents (bootfiles)`
which i can see is a more recent change, by design: [https://rauc.readthedocs.io/en/v1.12/faq.html#why-does-bundle-creation-fail-with-a-not-supported-as-contents-error](docs)

My issue is that it is not clear to me how to proceed from here. I can see the bootfiles folder and its contents being added to the tmp/<rauc-layer>/bundle/ folder, but building fails with the above error.
I would like to end up with a similar bundle structure as before.

ejoerns, 2025-01-04:
@mortenboye An option would be to not use a subdirectory for these files but to put them plain into the bundle.

However, `RAUC_BUNDLE_EXTRA_FILES` currently does not support adding files from a deploy dir subdirectory like `bootfiles` this way. It explicitly creates the subdirectories inside the bundle. This might need to be changed if RAUC sticks to the current behavior.

How do you update the bootfiles in your post-install to ensure the update is fail-safe / atomic?

If the current bundle structure is mandatory for you, adding your use case to the discussion in https://github.com/rauc/rauc/discussions/1533 might make sense.

Thomas-at, 2025-01-16:
Hello, I am also struggling with this exact behavior. Similar to mortenboye I am also trying to add multiple files from a directory inside the deploy directory to the bundle. I am not trying to add a directory itself or read from one recursively,  just adding files from a dir.
`RAUC_BUNDLE_EXTRA_FILES += "boot_scripts/file1 boot_scripts/file2"`

Putting the files directly into the deploy dir (without the boot_scripts sub dir) works, but would quite clutter the deploy directory itself.

Simply allowing a path to a file would help.

b4u-mw, 2025-07-03:
I came up with this solution. It's ugly like hell but works.

In update-bundle.bbappend:
```
# Add bash dependency for the hook script
RDEPENDS:${PN} += "bash"

# Add config.txt to RAUC bundle
RAUC_BUNDLE_EXTRA_DEPENDS += "rpi-config"

# Copy config.txt to the root of DEPLOY_DIR_IMAGE for RAUC bundle inclusion
do_configure:prepend() {
    import shutil
    DEPLOY_DIR_IMAGE = d.getVar('DEPLOY_DIR_IMAGE')
    BOOTFILES_DIR_NAME = d.getVar('BOOTFILES_DIR_NAME')
    shutil.copy(f"{DEPLOY_DIR_IMAGE}/{BOOTFILES_DIR_NAME}/config.txt", f"{DEPLOY_DIR_IMAGE}/config.txt")
}

do_configure:append() {
    import os
    os.remove(f"{DEPLOY_DIR_IMAGE}/config.txt")
}

RAUC_BUNDLE_EXTRA_FILES += "config.txt"

```